moveit/moveit_tutorials #5: Overall documentation structure

uwesterh, 2016-08-23:
Currently all pages start with a detailed discussion of the source code after only a very short introduction. The link to the whole file (section "The entire code") and the how to run sections are provided only at the end of the document which is quite confusing when reading the document the first time.

I suggest to move the "The entire code" section to the top of the file (after the introduction) and to insert a link to the "Running the code" section in the introduction or the "The entire code" section.

davetcoleman, 2016-08-23:
that's good feedback - I'll merge such a PR!

davetcoleman, 2018-10-13:
@uwesterh we had a large refactor of the tutorials this year, lead by @mlautman. We discussed how to best structure the tutorials and I think this issue is now out of date. Feel free to re-open if needed.

closed 2018-10-13 by davetcoleman