moveit/moveit_tutorials #15: Moveit for baxter

mjm522, 2016-09-24:
Hi, I recently started using moveit for a experiment setup with Baxter robot in simulation. I set up config files for baxter following your PR2 example. It successfully created it and I am able to plan using RVIZ (using the interface 'Plan-execute for random feasible links'). Though the arm moves, the RVIZ terminal window gets accumulated with following errors.

```
[ERROR] [1474740644.022261902, 6597.414000000]: Joint 'fl_caster_l_wheel_joint' not found in model 'baxter'
[ERROR] [1474740644.022289410, 6597.414000000]: Joint 'fl_caster_r_wheel_joint' not found in model 'baxter'
[ERROR] [1474740644.022321494, 6597.414000000]: Joint 'fr_caster_rotation_joint' not found in model 'baxter'
[ERROR] [1474740644.022342243, 6597.414000000]: Joint 'fr_caster_l_wheel_joint' not found in model 'baxter'
[ERROR] [1474740644.022356311, 6597.414000000]: Joint 'fr_caster_r_wheel_joint' not found in model 'baxter'
```
 
There are quite so many of them. But the interesting thing is that none of these joints are present in the baxter urdf file. I don't know from where it picks them up. I am using KDL as the inverse kinematics library.

Same issue happens when I try to write the python script and later it crashes with the following message:
```
terminate called after throwing an instance of 'class_loader::LibraryUnloadException'
  what():  Attempt to unload library that class_loader is unaware of.
Aborted (core dumped) 
```
Could you please suggest something on this?
Thanks in advance.

davetcoleman, 2016-09-25:
You are somehow still using configuration files, roslaunch files, etc from the PR2 tutorials. Grep through your configuration files to figure out where its coming from. Also try restarting roscore

mjm522, 2016-09-25:
restarting roscore helped. :)

closed 2016-09-25 by mjm522