moveit/moveit_tutorials #21: package failing with kinetic and ubuntu 16.04

arpit15, 2016-10-01:
I tried to build the package with the following commands which lead to errors and unsuccessful build with ros kinetic and ubuntu 16.04
`cd ~/catkin_ws/src`
`git clone https://github.com/ros-planning/moveit_tutorials.git`
`cd ..`
`catkin build`

[error gist](https://gist.github.com/arpit15/36e83985d6d79dc4e424d601729850b1)

v4hn, 2016-10-01:
see #22 and #23 
In short: adding `add_compile_options(-std=c++11)` to the top-level `CMakeLists.txt` should fix the failure, but you also need to add #22 (which will probably be merged in a bit).

davetcoleman, 2016-10-20:
it would be ideal if a PR2 user/maintainer could work on getting moveit_tutorials building in kinetic (has lots of PR2 deps)

v4hn, 2016-12-09:
should work by now.

closed 2016-12-09 by v4hn

vashmata, 2018-11-14:
Would it be possible to update the moveit tutorials wiki to add this information somewhere? I (someone with not much experience) came across this error and spent a whole day trying to figure out what exactly was wrong.

What ended up working for me was [this link](https://answers.ros.org/question/307792/catkin-build-error-for-moveit-workspace-setup/) which told me to checkout the kinetic-devel branch, and what confuses me is that it seems like a totally different solution but it still worked...

davetcoleman, 2018-11-15:
@vashmata please be more specific what error you encountered. The tutorials have been dramatically improved since 2016 when this issue was created. Please open a new issue with specific details about what was misleading. Even better, create a pull request yourself (its easy!) to make your suggested improvements. 

Thanks for your help!