moveit/moveit_tutorials #37: Please add some clarifying info to MoveGroup tutorial

mcevoyandy, 2016-10-23:
I find these lines extremely confusing. Could someone could unpack them into something more readable / understandable? Maybe the first is a really efficient way to do it, but it doesn't seem appropriate for a tutorial. JointModelGroup and RobotModel aren't discussed at all so the tutorial also seems very incomplete.

offending lines:
https://github.com/ros-planning/moveit_tutorials/blob/kinetic-devel/doc/pr2_tutorials/planning/src/move_group_interface_tutorial.cpp#L145

https://github.com/ros-planning/moveit_tutorials/blob/kinetic-devel/doc/pr2_tutorials/planning/src/move_group_interface_tutorial.cpp#L187

davetcoleman, 2016-10-24:
I can address this after this is merged: https://github.com/ros-planning/moveit_tutorials/pull/38

davetcoleman, 2016-10-28:
https://github.com/ros-planning/moveit_tutorials/pull/42

I've cleaned up that tutorial, though I'm not sure what to say about the [axis constraints](https://github.com/ros-planning/moveit_tutorials/blob/indigo-devel/doc/pr2_tutorials/planning/src/move_group_interface_tutorial.cpp#L167) because I haven't used them.

closed 2016-10-28 by davetcoleman