moveit/moveit_tutorials #40: Add Google Analytics tracking

davetcoleman, 2016-10-26:
This might be a good reference: http://perso.crans.org/besson/ga.en.html

Our tracking code:

```
<script>
  (function(i,s,o,g,r,a,m){i['GoogleAnalyticsObject']=r;i[r]=i[r]||function(){
  (i[r].q=i[r].q||[]).push(arguments)},i[r].l=1*new Date();a=s.createElement(o),
  m=s.getElementsByTagName(o)[0];a.async=1;a.src=g;m.parentNode.insertBefore(a,m)
  })(window,document,'script','https://www.google-analytics.com/analytics.js','ga');

  ga('create', 'UA-33626765-1', 'auto');
  ga('send', 'pageview');

</script>
```

v4hn, 2016-10-27:
Actually, I would prefer to *remove* external dependencies on cloud services instead of adding more of them.
I don't see the need to use GA on the MoveIt websites and it keeps bugging me that our *really simple* menu bar on `moveit.ros.org` requires `ajax.googleapi.com`.

davetcoleman, 2016-10-27:
Pre-grad school I was a web developer... GA is the standard for tracking usage. We've been using it since the beginning of the moveit project and I don't see why we should switch to some hard to maintain custom thing. Because every website uses the same GA, the scripts are almost always pre-cached in your browser and on nearby internet cache servers. We should stick to web standards...

v4hn, 2016-10-28:
I do have a strong opinion on this matter but I feel that should not influence MoveIt here.
So I'll just keep staying away from design decisions for the website and leave that to the others..

davetcoleman, 2016-10-28:
Having website usage statistics is nice to understand MoveIt!'s health and popularity. Using Google Analytics is the easiest and most common way to do that. I am assuming you distrust Google? If you want to spear head setting up some alternative that has similar functionality I won't mind, but I am assuming you don't actually want to bother with this, in which case I think we should take the easiest route. 

In addition, it is nice that we already have lots of historical data in GA.