moveit/moveit_tutorials #46: fake controllers under indigo

danielhoeltgen, 2016-11-02:
The fake controllers only jump to the last state and do not interpolate as described in [this tutorial](http://docs.ros.org/indigo/api/moveit_tutorials/html/doc/fake_controller_manager_tutorial.html).
The problem seems to be fixed under kinetic as discussed [here](https://groups.google.com/forum/#!topic/moveit-users/S-Cve5DFMS4)

rhaschke, 2016-11-02:
The corresponding changes were merged into Indigo with https://github.com/ros-planning/moveit/commit/21fd7e2e6414a8bcb39091ba1cc8fcfdfd2169e4.
However, there wasn't a release since then. Did you build from source?

danielhoeltgen, 2016-11-03:
Great, I build it from source and now it works. There is still one problem: "rate" or "fake_interpolating_controlller_rate" do not seem to have any effect. The trajectory of the interpolate type is not smooth even for high rate values. Should this work or is it not implemented so far?

rhaschke, 2017-01-10:
The fake controllers publish their joint state via `/move_group/fake_controller_joint_states` that are republished by a `joint_state_publisher` to `/joint_states` and finally received by MoveIt and other tools again.
Along that pipeline, there are several point of throttling, which effectively limits the maximal rate.

v4hn, 2017-01-12:
Closing this. The controllers are available in indigo and the rate is limited by design limitations.

closed 2017-01-12 by v4hn