moveit/moveit_tutorials #54: Task Space Goals

lakehanne, 2016-12-09:
Hi there,

Thank you for the tutorial. Is there an existing class method in the `robot_state` class that supports working  in the task space of the robot?
The documentation doesn't say anything about it.

v4hn, 2016-12-09:
Are you looking for `RobotState::setFromIK` and `RobotState::computeCartesianPath`?

lakehanne, 2016-12-09:
Thanks! Where is the documentation for this please?

v4hn, 2016-12-09:
They are mentioned in some places in the tutorials:
```
$ grep -r setFromIK
doc/pr2_tutorials/interactivity/src/interactive_robot.cpp:  if (robot_state_->setFromIK(group_, desired_group_end_link_pose_, 10, 0.1))
doc/pr2_tutorials/state_display/src/state_display_tutorial.cpp:    bool found_ik = kinematic_state->setFromIK(joint_model_group, end_effector_pose, 10, 0.1);
doc/pr2_tutorials/planning/src/move_group_interface_tutorial.cpp:  start_state.setFromIK(joint_model_group, start_pose2);
doc/pr2_tutorials/kinematics/src/kinematic_model_tutorial.cpp:  bool found_ik = kinematic_state->setFromIK(joint_model_group, end_effector_state, 10, 0.1);
```

```
$ grep -r computeCartesian
doc/pr2_tutorials/planning/src/move_group_interface_tutorial.cpp:  double fraction = move_group.computeCartesianPath(waypoints, eef_step, jump_threshold, trajectory);
```

The API documentation of the `RobotState` is [here](http://docs.ros.org/indigo/api/moveit_core/html/classmoveit_1_1core_1_1RobotState.html)

The rest is in the header files and the source itself. After all this is open source.

I'll close this as this is not about an issue with the framework.
If you need further advice, feel free to file questions on http://answers.ros.org

closed 2016-12-09 by v4hn