moveit/moveit_tutorials #55: How to get the goal state of Rviz by programming?

LMNcnu, 2017-01-03:
In the Rviz, I drag the marker on an arm of robot, then the orange color represent the goal state. How to get the goal state of Rviz by programming?
[image]

davetcoleman, 2017-01-03:
Create your own [interactive markers](http://wiki.ros.org/interactive_markers) and connect to an IK solver using custom code. Or use my pre-made version available in [moveit_visual_tools](https://github.com/ros-planning/moveit_visual_tools/blob/kinetic-devel/include/moveit_visual_tools/imarker_robot_state.h)

closed 2017-01-03 by davetcoleman

LMNcnu, 2017-01-05:
Hi，I want to get the orange arm's state(just every joint's angle), could you tell me?
Because the orange's arm is goal state ，is not current state. So I can't get the state by move_group.getCurrentState().

davetcoleman, 2017-01-08:
You will need to create your own solution (see above comment of mine) because the GUI you are referring to is not intended to be used along with custom code.

LMNcnu, 2017-01-09:
OK，thanks