moveit/moveit_tutorials #61: build fails on Indigo and Ubuntu 14.04

MarqRazz, 2017-01-11:
I am getting build errors with a fresh copy of the tutorial package. I'm running Indigo on Ubuntu 14.04 and made sure my system is up to date.

`cd ~/moveitEx/src`
`cd moveitEx/src`
`catkin_init_workspace`
`git clone https://github.com/ros-planning/moveit_tutorials.git`
`cd ..`
`catkin build`

Build Error...
`/home/marq/moveitEx/src/moveit_tutorials/doc/pr2_tutorials/pick_place/src/pick_place_tutorial.cpp:41:62: fatal error: moveit/move_group_interface/move_group_interface.h: No such file or directory
 #include <moveit/move_group_interface/move_group_interface.h>`

I get the same error for the pick place tutorial. I tried adding `add_compile_options(-std=c++11)` to the top-level CMakeLists.txt but I still get the same error. [#23](https://github.com/ros-planning/moveit_tutorials/issues/23)

MarqRazz, 2017-01-11:
This was my fault. I was pointed at the kinect-devel branch. After pointing to the correct branch the build was successful.

closed 2017-01-11 by MarqRazz

yxd5208, 2017-07-20:
Hello, I ran into the same problem. What did you need to do to point at the correct branch?

MarqRazz, 2017-07-20:
cd to the package and you can run the following commands.

git branch -a  //List all branches
git checkout <name_of_branch>  //Checkout the branch that you want
  git checkout indigo-devel

yxd5208, 2017-07-20:
That worked perfectly. Thank you very much for your help!