moveit/moveit_tutorials #67: lib version so.0.9.4 is not available for kinect unbuntu

MinMengbin, 2017-02-28:
/usr/bin/ld: warning: libmoveit_trajectory_execution_manager.so.0.9.4, needed by /opt/ros/kinetic/lib/libmoveit_move_group_interface.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_planning_scene_monitor.so.0.9.4, needed by /opt/ros/kinetic/lib/libmoveit_move_group_interface.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_model.so.0.9.4, needed by /opt/ros/kinetic/lib/libmoveit_move_group_interface.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_transforms.so.0.9.4, needed by /opt/ros/kinetic/lib/libmoveit_move_group_interface.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_state.so.0.9.4, needed by /opt/ros/kinetic/lib/libmoveit_move_group_interface.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_kinematic_constraints.so.0.9.4, needed by /opt/ros/kinetic/lib/libmoveit_move_group_interface.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_model_loader.so.0.9.4, needed by /opt/ros/kinetic/lib/libmoveit_common_planning_interface_objects.so, not found (try using -rpath or -rpath-link)
/opt/ros/kinetic/lib/libmoveit_move_group_interface.so: undefined reference to `planning_scene_monitor::CurrentStateMonitor::waitForCurrentState(ros::Time, double) const'
collect2: error: ld returned 1 exit status
myworkcell_core/CMakeFiles/myworkcell_node.dir/build.make:180: recipe for target '/home/aspa1/catkin_ws/devel/lib/myworkcell_core/myworkcell_node' failed
make[2]: *** [/home/aspa1/catkin_ws/devel/lib/myworkcell_core/myworkcell_node] Error 1
CMakeFiles/Makefile2:7956: recipe for target 'myworkcell_core/CMakeFiles/myworkcell_node.dir/all' failed
make[1]: *** [myworkcell_core/CMakeFiles/myworkcell_node.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Hi, I want to build a new node from my project by using moveit_group class to control IRB120 rabb. But When I use the function like 
moveit::planning_interface::MoveGroupInterface move_group(PLANNING_GROUP); from header of
#include <moveit/move_group_interfacemove_group_interface.h>  
it could not pass the catkin_make.  It looked like the version of the library was not right. I got lost. Cou you help me? Thanks!

v4hn, 2017-03-01:
Are you sure your system is up to date?
I believe updates should resolve this problem.

MinMengbin, 2017-03-01:
Hello, thanks for your reply. First I update the packages, I saw the following output messages. And it works. Thanks!

Setting up ros-kinetic-moveit-core (0.9.4-0xenial-20170213-222630-0800) ...
Setting up ros-kinetic-moveit-ros-perception (0.9.4-0xenial-20170213-224151-0800) ...
Setting up ros-kinetic-moveit-ros-planning (0.9.4-0xenial-20170213-225943-0800) ...
Setting up ros-kinetic-moveit-ros-robot-interaction (0.9.4-0xenial-20170213-231543-0800) ...

closed 2017-03-01 by v4hn