moveit/moveit_tutorials #72: move_gruop_interface_tutorial

japalmer29, 2017-04-20:
I installed the moveit_tutorials using the git clone command ... however when I try to run:

roslaunch moveit_tutorials move_group_interface_tutorial.launch

as the movegroup tutorial suggests, there is an error that no launch file exists. And there is no launch directory in moveit_tutorials, so that makes sense. Is this tutorial up to date? I'm using ros kinetic. Thanks.

v4hn, 2017-04-20:
The launch files are in the doc subfolder: https://github.com/ros-planning/moveit_tutorials/tree/kinetic-devel/doc/pr2_tutorials/planning/launch
Are you sure you build the workspace?

japalmer29, 2017-04-20:
When I try to run catkin_make for the moveit_tutorials package, I get the following errors about missing class object members and build failure:

[ 94%] Building CXX object moveit/moveit_ros/visualization/robot_state_rviz_plugin/CMakeFiles/moveit_robot_state_rviz_plugin_core.dir/moveit_robot_state_rviz_plugin_core_automoc.cpp.o
/home/bmi/Documents/pr2_ws/src/moveit_tutorials/doc/pr2_tutorials/planning/src/move_group_interface_tutorial.cpp: In function ‘int main(int, char**)’:
/home/bmi/Documents/pr2_ws/src/moveit_tutorials/doc/pr2_tutorials/planning/src/move_group_interface_tutorial.cpp:88:16: error: ‘class moveit_visual_tools::MoveItVisualTools’ has no member named ‘loadRemoteControl’
visual_tools.loadRemoteControl();
^
/home/bmi/Documents/pr2_ws/src/moveit_tutorials/doc/pr2_tutorials/planning/src/move_group_interface_tutorial.cpp:96:16: error: ‘class moveit_visual_tools::MoveItVisualTools’ has no member named ‘trigger’
visual_tools.trigger();
^
.... MORE MISSING MEMBER ERRORS .....

[ 94%] Linking CXX executable /home/bmi/Documents/pr2_ws/devel/lib/moveit_tutorials/attached_body_tutorial
moveit_tutorials/doc/pr2_tutorials/planning/CMakeFiles/move_group_interface_tutorial.dir/build.make:62: recipe for target 'moveit_tutorials/doc/pr2_tutorials/planning/CMakeFiles/move_group_interface_tutorial.dir/src/move_group_interface_tutorial.cpp.o' failed
make[2]: *** [moveit_tutorials/doc/pr2_tutorials/planning/CMakeFiles/move_group_interface_tutorial.dir/src/move_group_interface_tutorial.cpp.o] Error 1
CMakeFiles/Makefile2:24683: recipe for target 'moveit_tutorials/doc/pr2_tutorials/planning/CMakeFiles/move_group_interface_tutorial.dir/all' failed
make[1]: *** [moveit_tutorials/doc/pr2_tutorials/planning/CMakeFiles/move_group_interface_tutorial.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 94%] Built target attached_body_tutorial
[ 94%] Linking CXX executable /home/bmi/Documents/pr2_ws/devel/lib/moveit_tutorials/pick_place_tutorial
[ 94%] Built target pick_place_tutorial
[ 94%] Linking CXX shared library /home/bmi/Documents/pr2_ws/devel/lib/libmoveit_robot_state_rviz_plugin_core.so
[ 94%] Linking CXX shared library /home/bmi/Documents/pr2_ws/devel/lib/libmoveit_planning_scene_rviz_plugin_core.so
[ 94%] Built target moveit_robot_state_rviz_plugin_core
[ 94%] Built target moveit_planning_scene_rviz_plugin_core
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

japalmer29, 2017-04-20:
The pr2_moveit_config package builds ok, but I get an error when I roslaunch pr2_moveit_config demo.launch, in RViz:

The class required for this panel, 'rviz_visual_tools/RvizVisualToolsGui', could not be loaded.
Error:
According to the loaded plugin descriptions the class rviz_visual_tools/RvizVisualToolsGui with base class type rviz::Panel does not exist. Declared types are rviz_plugin_tutorials/Teleop

And motion planning, e.g. QueryStartState, etc. doesn:t work.

There are also a bunch of the following error messages in the terminal window after roslaunch:

TIFFFieldWithTag: Internal error, unknown tag 0xa402.

davetcoleman, 2017-05-12:
>  get an error when I roslaunch pr2_moveit_config demo.launch

What version of rviz_visual_tools are you using? kinetic-devel from source? are you on indigo?

> TIFFFieldWithTag: Internal error, unknown tag 0xa402.

That is a PR2, reported here: https://github.com/PR2/pr2_common/issues/257https://github.com/PR2/pr2_common/issues/257

japalmer29, 2017-05-14:
I'm using kintitc. I'm installed from the "kinetic workaround" instructions on the tutorial page.

davetcoleman, 2017-06-05:
Perhaps re-source your workspace and relaunch rviz?

sidharthrajaram, 2017-07-12:
I'm having the same issue as @japalmer29. Set up works fine but when "roslaunch pr2_moveit_config demo.launch" is run, I get an initial set of warnings that say 

**"[ WARN] [1499889173.721194396]: The root link base_footprint has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF."**

and then a neverending stream of error messages

**TIFFFieldWithTag: Internal error, unknown tag 0xa402.**
and occasionally in the middle of the stream, this message appears:
**TIFFFetchNormalTag: Warning, Incompatible type for "RichTIFFIPTC"; tag ignored.**

Did anyone resolve it yet?
Thanks.

davetcoleman, 2017-07-21:
Those are problems with the [pr2_common](https://github.com/PR2/pr2_common) repo - please file bug reports or fixes there for those warnings.

felixvd, 2021-02-10:
I'll close this as it seems to be a bug with pr2_common, if anything.

closed 2021-02-10 by felixvd