moveit/moveit_tutorials #73: rvis visual tools

japalmer29, 2017-04-20:
When I launch pr2_moveit_config demo.launch, I get an error in RViz saying rviz_visual_tools class not loaded. And I can't manipulate or plan anything with the robot. I have moveit_visual_tools. I'm not sure how to install any of these demos. I had to run "cmake ." in pr2_moveit_config and then source devel/setup.bash in order to get the demo.launch to run. But none of this is explained in the tutorial. It just says run git clone ... and then roslaunch ...., but that doesn't work.

v4hn, 2017-04-20:
Your problems seem to boil down to missing knowledge about ROS workspaces and catkin.
This is supposed to be basic knowledge when using MoveIt, so it's not in our instructions, but in the official ROS tutorials at http://wiki.ros.org

japalmer29, 2017-04-20:
I have gone through the ros tutorials. Unfortunately they are not very helpful. E.g http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment gives practically no useful information on dealing with the moveit packages or workspaces.

v4hn, 2017-04-20:
There are more tutorials there. In particular two steps later: http://wiki.ros.org/ROS/Tutorials/CreatingPackage

japalmer29, 2017-04-20:
Thanks, I saw that one. But I don't know what the dependencies should be for catkin_create_pkg. The same as the example? Is it necessary to run cmake on the packages? Do I just run "cmake ." in the folder? I don't see that in the tutorials.

v4hn, 2017-04-20:
From the Tutorial I referenced above:

    4. Building a catkin workspace and sourcing the setup file
 
    Now you need to build the packages in the catkin workspace:
    
    $ cd ~/catkin_ws
    $ catkin_make
    
    After the workspace has been built it has created a similar structure in the devel subfolder as you usually find under /opt/ros/$ROSDISTRO_NAME.
    
    To add the workspace to your ROS environment you need to source the generated setup file:
    
    $ . ~/catkin_ws/devel/setup.bash

You don't have to create a package, because the tutorial repository you cloned already *is* the package you want to build

japalmer29, 2017-04-20:
Ok, thanks, it's clearer now. But I am getting an error when trying to build using catkin_make:

[ 94%] Building CXX object moveit/moveit_ros/visualization/robot_state_rviz_plugin/CMakeFiles/moveit_robot_state_rviz_plugin_core.dir/moveit_robot_state_rviz_plugin_core_automoc.cpp.o
/home/bmi/Documents/pr2_ws/src/moveit_tutorials/doc/pr2_tutorials/planning/src/move_group_interface_tutorial.cpp: In function ‘int main(int, char**)’:
/home/bmi/Documents/pr2_ws/src/moveit_tutorials/doc/pr2_tutorials/planning/src/move_group_interface_tutorial.cpp:88:16: error: ‘class moveit_visual_tools::MoveItVisualTools’ has no member named ‘loadRemoteControl’
   visual_tools.loadRemoteControl();
                ^
/home/bmi/Documents/pr2_ws/src/moveit_tutorials/doc/pr2_tutorials/planning/src/move_group_interface_tutorial.cpp:96:16: error: ‘class moveit_visual_tools::MoveItVisualTools’ has no member named ‘trigger’
   visual_tools.trigger();
                ^
.... MORE MISSING MEMBER ERRORS .....

[ 94%] Linking CXX executable /home/bmi/Documents/pr2_ws/devel/lib/moveit_tutorials/attached_body_tutorial
moveit_tutorials/doc/pr2_tutorials/planning/CMakeFiles/move_group_interface_tutorial.dir/build.make:62: recipe for target 'moveit_tutorials/doc/pr2_tutorials/planning/CMakeFiles/move_group_interface_tutorial.dir/src/move_group_interface_tutorial.cpp.o' failed
make[2]: *** [moveit_tutorials/doc/pr2_tutorials/planning/CMakeFiles/move_group_interface_tutorial.dir/src/move_group_interface_tutorial.cpp.o] Error 1
CMakeFiles/Makefile2:24683: recipe for target 'moveit_tutorials/doc/pr2_tutorials/planning/CMakeFiles/move_group_interface_tutorial.dir/all' failed
make[1]: *** [moveit_tutorials/doc/pr2_tutorials/planning/CMakeFiles/move_group_interface_tutorial.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 94%] Built target attached_body_tutorial
[ 94%] Linking CXX executable /home/bmi/Documents/pr2_ws/devel/lib/moveit_tutorials/pick_place_tutorial
[ 94%] Built target pick_place_tutorial
[ 94%] Linking CXX shared library /home/bmi/Documents/pr2_ws/devel/lib/libmoveit_robot_state_rviz_plugin_core.so
[ 94%] Linking CXX shared library /home/bmi/Documents/pr2_ws/devel/lib/libmoveit_planning_scene_rviz_plugin_core.so
[ 94%] Built target moveit_robot_state_rviz_plugin_core
[ 94%] Built target moveit_planning_scene_rviz_plugin_core
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

Do you know why I am getting missing member errors about "trigger" etc., and why the build is failing? Thanks for you help.

japalmer29, 2017-04-20:
I also still get an error when I launch pr2_moveit_config demo.launch in RViz:

The class required for this panel, 'rviz_visual_tools/RvizVisualToolsGui', could not be loaded.
Error:
According to the loaded plugin descriptions the class rviz_visual_tools/RvizVisualToolsGui with base class type rviz::Panel does not exist. Declared types are rviz_plugin_tutorials/Teleop

And Robot planning, e.g. QueryStartState, etc. doesn:t work.

Also there are a bunch of the following error messages in the terminal window after roslaunch:

TIFFFieldWithTag: Internal error, unknown tag 0xa402.

v4hn, 2017-04-20:
Did you checkout the right branches of the repositories you cloned?
Looks like inconsistencies between indigo/kinetic to me.

japalmer29, 2017-04-20:
I followed the git clone instructions on the page: http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/move_group_interface_tutorial.html

which seem to refer to kinetic, which is what I have.

v4hn, 2017-04-20:
@davetcoleman wrote the visual tools and the integration code into the tutorials, so he can probably help better :)

japalmer29, 2017-04-20:
Ok, thanks for the assistance.

davetcoleman, 2017-05-13:
I ran through the tutorial just now and everything works fine on ROS Kinetic / Ubuntu 16.04. I've clarified the catkin setup instructions that might trip up new users: https://github.com/ros-planning/moveit_tutorials/pull/76

closed 2017-05-13 by v4hn

Hariscorner, 2017-10-04:
Hi all, 
I'm facing the same issue as @japalmer29 and i came through the same path as he did. In particular, the error that I'm receiving is :

**_The class required for this panel, 'rviz_visual_tools/RvizVisualToolsGui', could not be loaded.
Error:
According to the loaded plugin descriptions the class rviz_visual_tools/RvizVisualToolsGui with base class type rviz::Panel does not exist. Declared types are rviz_plugin_tutorials/Teleop_**

(I'm new to MoveIt! All I did so far was to go through this [tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/ros_visualization/visualization_tutorial.html) and the corresponding setup assistant tutorial linked there)

Is there a fix for the issue? Thanks in advance.

davetcoleman, 2017-10-09:
I imagine you are not sourcing your workspace correctly, or not downloading the required dependencies using ``rosdep install``

anubhav1772, 2020-03-03:
I was getting the same error in kinetic, but doing `sudo apt-get install ros-kinetic-rviz-visual-tools` help me get rid of the error.

irenjie, 2020-11-18:
> I was getting the same error in kinetic, but doing `sudo apt-get install ros-kinetic-rviz-visual-tools` help me get rid of the error.

helpful for me

siddarth09, 2021-11-12:
also try `rosdep  install --from-paths src --ignore-src -r -y`
 this solved the error for me