moveit/moveit_tutorials #74: Setup Assistant Tutorial Questions

mcevoyandy, 2017-04-26:
Could you add more detail to `Section 4: Add Planning Groups` and `Section 6: Label End Effectors`?

Section 6 says end-effectors are special groups that allows special operations to happen on them internally. It would be helpful to elaborate on these special operations or point the reader to more detailed info.

My questions in section 4 stem from this lack of understanding. Following the Move Group tutorial the goal pose is for the `wrist_roll_joint` so in an actual situation you'd always want to plan to some offset pose. What's the advantage of this versus putting a dummy link in there so you always plan to the gripper "center"?

Is it possible to add a tutorial on gripping so that maybe some of these issues become clear?

davetcoleman, 2017-04-27:
I think there was more intended to be "special" for those groups than was ever implemented. For inverse kinematics, in most locations in the code if no end effector is specified, it will still default to using the last link in a chain to act as a fake end effector. For example, these EE groups are used for choosing where to put the interactive marker in the Rviz plugin. 

Using a dummy link within a URDF is sometimes done, but is semantically incorrect - the URDF is intended only to describe the actual geometry and physics of the robot. The SRDF is suppose to contain more "semantic" data such as tooling information.

A tutorial on gripping is sorely needed, somewhat related: https://github.com/ros-planning/moveit/issues/493

v4hn, 2017-04-27:
Yes, the current documentation of the grasping pipeline is not good.
Also the pipeline itself has a number of flaws you will see if you look into the implementation,
so there is quite a bit of work left to do there.

The EndEffector definition is used in at least three situations:
- To figure out where to add InteractiveMarkers in the RViz display
- To connect the planning group of a gripper and the one of the arm holding it, so that you don't always have to specify both
- To define one unique link for the gripper where grasped objects are attached to and which "mostly" defines the pose of your gripper for grasping. For historical reasons this is the `parent_link`.

The last two points might be somewhat entangled in the code to make the move_group behave "in a smart way" when the user requests something without fully specifying the details.
The situation could definitely be improved..

> My questions in section 4 stem from this lack of understanding. Following the Move Group tutorial the goal pose is for the wrist_roll_joint so in an actual situation you'd always want to plan to [...]

This offset is quite annoying and makes things harder to understand, I agree.
Feel free to declare your kinematic chain from base to a virtual link at the center of the end-effector instead.
This works and we use this over here in our lab in multiple places.

Also, this is somewhat unrelated to the EndEffector definition: you can define your `arm` group as
chain `base_link->arm_tool0->gripper_base->gripper_tool0` and have the planning group of your gripper include `gripper_base->gripper_tool0` and still make the end-effector definition of your gripper point to `arm` as parent_group.
The question is what the `parent_link` should be in this case.

In one of our setups [we use `gripper_tool0` for that](https://github.com/TAMS-Group/tams_ur5_setup/blob/master/tams_ur5_setup_moveit_config/config/tams_ur5_setup.srdf#L124) and we do successful grasping with the Pick pipeline with this one.
I know there is an offset bug in the place code though, that makes the setup fail to perform placing though.

candidk, 2017-10-18:
I attempted Setup Assistant Tutorial followed by the RViz Plugin Tutorial
No interactive marker was displayed in RViz and the following error was logged to terminal:

```
[ INFO] [1508344563.774111009]: Loading robot model 'pr2'...
[ERROR] [1508344563.785211118]: Group 'right_arm' is not a chain
[ERROR] [1508344563.785262245]: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'right_arm'
[ERROR] [1508344563.785979936]: Kinematics solver could not be instantiated for joint group right_arm.
[ INFO] [1508344563.799430428]: Loading robot model 'pr2'...
[ERROR] [1508344563.810831186]: Group 'left_arm' is not a chain
[ERROR] [1508344563.810856901]: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'left_arm'
[ERROR] [1508344563.811354335]: Kinematics solver could not be instantiated for joint group left_arm.

```
I solved this by modifying MoveIt Setup Assistant Planning Groups for right_arm (and left_arm) to be a chain instead of joints.

right_arm > Chain > r_shoulder_pan_link -> r_wrist_roll_link
right_gripper > Links > r_gripper_palm_link -> r_gripper_tool_frame
left_arm > Chain > l_shoulder_pan_link -> l_wrist_roll_link
left_gripper > Links > l_gripper_palm_link -> l_gripper_tool_frame

Should we update the MoveIt Setup Assistant tutorial docs

davetcoleman, 2017-10-18:
Which pr2_moveit_config package did you use? I believe many of the tutorials currently use this version:

https://github.com/davetcoleman/pr2_moveit_config/blob/master/config/pr2.srdf#L16

naveedhd, 2017-12-04:
@davetcoleman the one generated by following [this tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/setup_assistant/setup_assistant_tutorial.html) is a bit different from [davetcoleman/pr2_moveit_config](https://github.com/davetcoleman/pr2_moveit_config). One difference, as @canktech mentioned, is arms specified as chain vs. set of joints. 

Other minor thing I mistook is no explicit mention of **Not** selecting a solver for left/right gripper.

Would you suggest modifying the setup_assitant_tutorial that it generates the same config (or atleast similar) that will suppress the errors?

samarth-robo, 2018-12-29:
@canktech I experienced the same while creating a MoveIt! config for KUKA IIWA from the [kuka_lbr_iiwa_support](https://github.com/ros-industrial/kuka_experimental/tree/indigo-devel/kuka_lbr_iiwa_support) package. I was following the [official MoveIt! Setup Assistant tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/setup_assistant/setup_assistant_tutorial.html).

Inspired by your comment, I modified the instructions in Step 4 of the tutorial. While adding my planning groups, I added them using the `Add Kin. Chain` button instead of the `Add Joints` or `Add Links` buttons. This solved the error and interactive markers were displayed correctly.

davetcoleman, 2019-01-02:
@samarth-robo could you update the tutorial to be clearer around this please? a quick PR would be awesome

thomascp, 2021-08-20:
Hi all,
I have a self-made dual arm robot.
I did a test on pr2, when I choosing arms, it will show two interactive marker for both arms.
But in my case, I don't have any, but if I choose right_arm or left_arm, there will be one interactive marker.
This is my srdf, quite similar to the pr2's.
```
    <group name="left_arm">
        <chain base_link="body_link" tip_link="left_arm_link4" />
    </group>
    <group name="right_arm">
        <chain base_link="body_link" tip_link="right_arm_link4" />
    </group>
    <group name="both_arms">
        <group name="left_arm" />
        <group name="right_arm" />
    </group>
```
I don't know how pr2 setup arms plan group, it would be very appreciated that someone can help me, thanks!

BR,
Peng