moveit/moveit_tutorials #75: Move Group Interface Tutorial

mcevoyandy, 2017-05-02:
In the `Move Group Interface Tutorial` the section `Moving to a pose goal` says you can use `move_group.move()` to move to a pose goal but "need a controller to be active and report success on execution of a trajectory."

Just for visualizations, how do you use `move_group.plan()` and `move_group.planCartesianPath()` in succession so that you can visualize a more elaborate set of motions? How do you "accept" the computed plan and use that as the start for another move?

My goal is to use the parts of this tutorial to:

1. plan to some point in space
2. attach an object
3. Cartesian move
4. plan to some new point in space

Froz3nArcher, 2017-05-03:
@mcevoyandy The tutorial appears to be misleading regarding the use of `move_group.plan()` followed 
 by `move_group.move()`.  The `move_group.move()` function does not appear to use the `Plan` created by `move_group.plan()`.  The documentation indicates that `move_group.execute()` uses a specified plan, but `move_group.move()` "plans and executes" the move.  This could explain the difference you see between the plan and the move.

In addition, there does not appear to be a `execute()` or `move()` function that could use the `RobotTrajectory` calculated in `computeCartesianPath()`.  Is the recommended usage to store the returned `RobotTrajectory` into a `Plan` and then call `move_group.execute()`?

v4hn, 2017-05-08:
@mcevoyandy It is not really clear to me what this issue is about.
"How do I use MoveIt to plan paths" is a bit too broad for a single Issue.

You can `plan` multiple trajectories, set their corresponding start state explicitly before planning and put the resulting plans into one Plan as @Froz3nArcher suggested and execute it.
You will have to ensure correct time parameterization yourself in this case.
You could also store the separate plans and execute them one after the other.

For cartesian trajectories you have to create the Plan yourself at the moment.
Feel free to add a pull request to the moveit repository that adds a method that returns a Plan though.

mcevoyandy, 2017-05-08:
I think @Froz3nArcher has cleared things up for me. From the tutorial it
wasn't clear what `plan` was actually doing and `execute` wasn't mentioned
at all. It was also unclear from the tutorial that you have to create the
plan yourself for Cartesian trajectories.

I guess the issue is adding language to the tutorial to clarify these
issues.

Andy McEvoy | (832) 439 - 6150

sheikspeare, 2019-11-20:
https://github.com/davetcoleman/moveit_tutorials/commit/3b53c756d2194e0b825dd48df875ba398d811590

davetcoleman, 2019-11-20:
Thanks @sheikspeare , closing this old issue.

closed 2019-11-20 by davetcoleman