moveit/moveit_tutorials #77: Clarifications for movegroup's go, plan, and compute_cartesian_path methods

scottpaulin, 2017-05-14:
I think this *might* have some crossover with #75, but is more specific. Nothing within the tutorial is wrong, it's actually very concise and easy to follow. A few extra lines could clarify the uses of the go, plan and compute_cartesian_path methods.

The tutorial covers using movegroup.go, movegroup.plan methods. The first part details how to compute a plan:

`plan1 = group.plan()`

And then a little later it says that the go() function can be used to move the actual robot to the pose goal:

```

  ## Moving to a pose goal
  ## ^^^^^^^^^^^^^^^^^^^^^
  ##
  ## Moving to a pose goal is similar to the step above
  ## except we now use the go() function. Note that
  ## the pose goal we had set earlier is still active 
  ## and so the robot will try to move to that goal. We will
  ## not use that function in this tutorial since it is 
  ## a blocking function and requires a controller to be active
  ## and report success on execution of a trajectory.

  # Uncomment below line when working with a real robot
  # group.go(wait=True)
```

Since the previous step of the tutorial covered how to compute a motion plan, one might expect that go() just executes this plan. Could we add one sentence to say that execute(plan1) can be used to make the robot follow plan1 but go() can be used to compute and follow a new plan?

Later in the tutorial the Cartesian path api is covered:

```
  (plan3, fraction) = group.compute_cartesian_path(
                               waypoints,   # waypoints to follow
                               0.01,        # eef_step
                               0.0) # jump_threshold
```

There is no code here to show how how we can make the robot follow the Cartesian path. Could we add one line here to state that group.execute(plan3) should be used to make the robot follow the computed Cartesian path?

davetcoleman, 2017-06-04:
Please simply propose a PR with your suggested clarifications and I don't see why we wouldn't merge it!

v4hn, 2017-06-07:
@scottpaulin Did you forget to create a request after your commit yesterday? :) The patch looks good to me :D

scottpaulin, 2017-06-07:
@v4hn yes *facepalm*. Thanks for the heads up, I would have never made the PR :p.

v4hn, 2017-06-19:
resolved in #78

closed 2017-06-19 by v4hn