moveit/moveit_tutorials #85: inconsistent namespace redefinitions for xmlns:xacro and RvizVisualToolsGui could not be loaded

yxd5208, 2017-07-20:
Hello, 
I recently installed this package. I'm using ROS indigo. When I launch demo.launch under pr2_tutorials or move_group_interface.launch, they all work. I can plan paths etc but I do get the following errors:
1. 
 ```
inconsistent namespace redefinitions for xmlns:xacro:
    old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
    new: http://ros.org/wiki/xacro (/opt/ros/indigo/share/pr2_description/urdf/torso_v0/torso.urdf.xacro)
```

2. 
`` PluginlibFactory: The plugin for class 'rviz_visual_tools/KeyTool' failed to load.  Error: According to the loaded plugin descriptions the class rviz_visual_tools/KeyTool with base class type rviz::Tool does not exist. Declared types are  rviz/FocusCamera rviz/Interact rviz/Measure rviz/MoveCamera rviz/PublishPoint rviz/Select rviz/SetGoal rviz/SetInitialPose rviz_plugin_tutorials/PlantFlag"

3.
`PluginlibFactory: The plugin for class 'rviz_visual_tools/RvizVisualToolsGui' failed to load.  Error: According to the loaded plugin descriptions the class rviz_visual_tools/RvizVisualToolsGui with base class type rviz::Panel does not exist. Declared types are  rviz_plugin_tutorials/Teleop`

[image]

I'm very new to ROS and MoveIt!, and I'd like to know what caused these problems. catkin_make doesn't return any errors and everything is compiled fine. 

I also made sure that my package is pointing to the right branch. This is the output of _git branch -a_:
[image]

v4hn, 2017-07-20:
The first error is related to the outdated pr2 description package and can be ignored.

regarding 2 and 3: Did you install the `ros-indigo-moveit-visual-tools` package or compile the package locally in your workspace?

yxd5208, 2017-07-22:
I believe so. I just ran 
`sudo apt-get install ros-indigo-moveit-visual-tools`
And received an output that says, "ros-indigo-moveit-visual-tools is already the newest version"
Thank you very much for your response :) 
Do you have any other advice about what might be going on?

yxd5208, 2017-07-22:
I tried to run the demo code: visual_tools_demo.launch in the package "moveit_visual_tools" and the random objects that are supposed to show up are not showing up at all. It seems like there's something wrong with that package. I uninstalled the package and reinstalled it and the problem is still not fixed.

davetcoleman, 2017-07-23:
the moveit_visual_tools features are only for Kinetic - if you use the indigo branch of moveit_tutorials you will notice moveit_visual_tools is not used. please consider upgrading to ROS Kinetic as it is an LTS release
http://docs.ros.org/indigo/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/move_group_interface_tutorial.html

closed 2017-07-23 by davetcoleman