moveit/moveit_tutorials #86: A problem of moveit_setup_assistant ! process has died.

wsycy1994, 2017-07-24:
when I execute "roslaunch moveit_setup_assistant setup_assistant.launch" ,I got the error like follow:
=========================================================================REQUIRED process [moveit_setup_assistant-2] has died!
process has died [pid 11143, exit code -11, cmd /opt/ros/indigo/lib/moveit_setup_assistant/moveit_setup_assistant __name:=moveit_setup_assistant __log:=/home/cy/.ros/log/01d2166e-7071-11e7-b681-c4346b72034a/moveit_setup_assistant-2.log].
log file: /home/cy/.ros/log/01d2166e-7071-11e7-b681-c4346b72034a/moveit_setup_assistant-2*.log
Initiating shutdown!
=========================================================================
[moveit_setup_assistant-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

please help me !!! thanks very much!!!!1

v4hn, 2017-07-24:
First guess: do you run the setup assistant inside VirtualBox et. al.?
There are known problems with the OpenGL context in this case.

Leaving that aside, you don't give us enough to go on.
If you want to figure out the problem, please build the setup assistant
in your own workspace with debug flags and run it through the gdb debugger.
If the segfault still occurs there, you can use `bt` to generate a
stack trace that includes a lot of information to help us debug this.

MaartenKleijwegt, 2018-01-03:
Ran across this problem as well. Running on Ubuntu 16.04 LTS using ROS kinetic. NO virtual machines were used. First of all, a lot of warnings are given due to using the dated PR2 model namely:

[rospack] Error: no package given
[librospack]: error while executing command
deprecated: xacro tags should be prepended with 'xacro' xml namespace.
Use the following script to fix incorrect usage:
        find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'
when processing file: /opt/ros/kinetic/share/pr2_description/urdf/common.xacro
included from: /opt/ros/kinetic/share/pr2_description/robots/pr2_no_kinect.urdf.xacro

inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/shoulder_v0/shoulder.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/upper_arm_v0/upper_arm.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/forearm_v0/forearm.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/gripper_v0/gripper.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/head_v0/head.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/tilting_laser_v0/tilting_laser.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/torso_v0/torso.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/base_v0/base.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/sensors/head_sensor_package.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/sensors/wge100_camera.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/sensors/projector_wg6802418.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/gazebo/gazebo.urdf.xacro)

after that there's a weird bug where the node fails to set the parameters initially due to not being able to contact the master:

[ INFO] [1515020473.376741972]: Loaded pr2 robot model.
[ INFO] [1515020473.377113356]: Setting Param Server with Robot Description
[ERROR] [1515020473.379147367]: [setParam] Failed to contact master at [localhost:11311].  Retrying...
[ INFO] [1515020473.453269674]: Connected to master at [localhost:11311]

Then there are some problems with the model, however, defaulting to some settings seems to fix that:

[ INFO] [1515020473.457769194]: Robot semantic model successfully loaded.
[ INFO] [1515020473.457823621]: Setting Param Server with Robot Semantic Description
[ INFO] [1515020473.507338059]: Loading robot model 'pr2'...
[ INFO] [1515020473.507382996]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1515020473.709099445]: Stereo is NOT SUPPORTED
[ INFO] [1515020473.709225877]: OpenGl version: 3 (GLSL 1.3).

and then the process suddenly dies:

================================================================================REQUIRED process [moveit_setup_assistant-1] has died!
process has died [pid 3391, exit code -11, cmd /opt/ros/kinetic/lib/moveit_setup_assistant/moveit_setup_assistant __name:=moveit_setup_assistant __log:=/home/maarten/.ros/log/05c3cb64-f0d7-11e7-b79a-54ee756d3429/moveit_setup_assistant-1.log].
log file: /home/maarten/.ros/log/05c3cb64-f0d7-11e7-b79a-54ee756d3429/moveit_setup_assistant-1*.log
Initiating shutdown!
================================================================================

looking at the logs referred to does not reveal any more info.  Will try to build the tutorials from github in the morning.

MaartenKleijwegt, 2018-01-05:
Compiling from source seems to fix the problem. However, there are still a lot of errors

v4hn, 2018-02-23:
I will close this as apparently fixed.

For new errors, please file different issues.

closed 2018-02-23 by v4hn