moveit/moveit_tutorials #89: Slight tweak to ros_visualization tutorial (Scene Robot -> Show Robot Visual)

jwhendy, 2017-07-31:
Hello,

Just going through the [tutorials]() and saw this bit:

>The display states for each of these visualizations can be toggled on and off using checkboxes.
>
>1. The start state using the "Query Start State" checkbox in the "Planning Request" tab.
>2. The goal state using the "Query Goal State" checkbox in the "Planning Request" tab.
>3. The planning scene robot using the "Show Scene Robot" checkbox in the "Scene Robot" tab.
>4. The planned path using the "Show Robot Visual" checkbox in the "Planned Path" tab.

If I'm correct that the third should be `Show Robot Visual`, I can submit a PR.

What I see in RViz:

[image]

v4hn, 2017-07-31:
Of course you are right. Could you file a pull-request that fixes it?

jwhendy, 2017-07-31:
Had it waiting :) Submitted. Let me know if you need any adjustments to it. Thanks!

closed 2017-08-01 by v4hn