moveit/moveit_tutorials #97: Help understanding general state of MoveIt and rospy

jwhendy, 2017-08-08:
Being new to ROS and going through tutorials (with python), I often find myself not knowing if some lesson is C++ specific, if it's just undocumented in python (but exists), and if the latter what the `rospy` analog is.

Having finished (and quite enjoyed) the [Move Group Python tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pr2_tutorials/planning/scripts/doc/move_group_python_interface_tutorial.html) , I looked to the tutorials on [Kinematic Models](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pr2_tutorials/kinematics/src/doc/kinematic_model_tutorial.html) and [Planning Senes](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/planning_scene_tutorial.html), finding they were only in C++.

I thought about trying to translate them based on my knowledge, but sort of fizzled out as I don't find much when I google things like "getJointModelGroup ROS python rospy" or similar.

Could someone help explain where things are at with `rospy`? Should I just assume that anything mentioned for C++ (with respect to MoveIt), exists in [moveit_commander](http://docs.ros.org/kinetic/api/moveit_commander/html/namespacemoveit__commander.html)?

For a couple of specific examples:

- is `getJointModelGroup` equal to [`get_current_joint_values`](http://docs.ros.org/kinetic/api/moveit_commander/html/classmoveit__commander_1_1move__group_1_1MoveGroupCommander.html#abe844b6e408d907b070e637c938da82b)?

- I don't see anything in the [`planning_scene_interface`](http://docs.ros.org/kinetic/api/moveit_commander/html/classmoveit__commander_1_1planning__scene__interface_1_1PlanningSceneInterface.html) that corresponds to `checkSelfCollision` mentioned [here](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/planning_scene_tutorial.html).

- generally speaking, I'm at the point where I'd like to take my knowledge and start doing things with a UR arm. What is the next step for this Padawan, such as complex motion paths/trajectories, pick and place, etc.? Is there some other resource or library that folks graduate to for more 'real world' tasks?

Lastly, if there *are* `rospy` equivalents to these sorts of things, I'd be happy to contribute tutorials for you. I'm going through it anyway and tend to like these sorts of things. Point me in the right direction and you'll get some work out of me!

philwall3, 2017-09-06:
I can agree with this. There is certainly a lot of stuff missing for Python.

jwhendy, did you already do some work regarding this?

jwhendy, 2017-09-06:
@philwall3 not at this point. My current project has not required me to go back to moveit yet. I will though, an would love to learn as well as contribute. If there was a definitive location for the information I could certainly try to "translate" some of the existing tutorials to python.

v4hn, 2017-09-06:
It is absolutely true that many parts of MoveIt are not exposed to python at the moment.
Additionally I believe there are a number of structures in MoveIt rospy that lack proper tutorials and/or documentation.

Patches that expose features missing in the Python API (if you require them), as well as patches for inline-documentation (in the source of the individual component) and tutorials in this repository are very much welcome and will be reviewed (hopefully) fast.

closed 2021-02-10 by felixvd