moveit/moveit_tutorials #98: moveit_tutorials planning_scene_ros_api_tutorial Object does not get removed. Moveit kinetic

Vonstafenberg, 2017-08-09:
@davetcoleman ,@v4hn

I am using Moveit with kinetic. I am not very expert with ROS and Moveit. Fortunately the contributors have updated some packages to make it work with kinetic. However there are some issues.
The Move_group_interface tutorial #45 works like shown in the video. But I have not tested all the tutorials. However there is a problem with moveit_tutorials planning_scene_ros_api_tutorial for pr2. The object appears and gets connected to the right gripper. But then it does not get detached. and along the way between the processes of attach and detach I get a stream of messages
.............................................................................

```
[ INFO] [1502275537.591629864]: Attached object 'box' to link 'r_wrist_roll_link'
[ INFO] [1502275537.652572290]: Detaching the object from the robot and returning it to the world.
[ERROR] [1502275537.653146469]: Found empty JointState message ``      **<-- also getting this**
```
```
[ERROR] [1502275537.653223449]: The specified RobotState is not marked as is_diff. The request to modify the object 'box' is not supported. Object is ignored.
[ INFO] [1502275537.815513782]: Attached object 'box' to link 'r_wrist_roll_link'
[ INFO] [1502275538.071848487]: Removing world object with the same name as newly attached object: 'box'
[ INFO] [1502275538.071963773]: Attached object 'box' to link 'r_wrist_roll_link'
[ INFO] [1502275539.831450395]: Attached object 'box' to link 'r_wrist_roll_link'
[ INFO] [1502275540.087312698]: Attached object 'box' to link 'r_wrist_roll_link'
[ INFO] [1502275540.343626912]: Attached object 'box' to link 'r_wrist_roll_link'
```
I have checked the same tutorial in **indigo**. it works inspite of
`[ERROR] [1502275537.653223449]: The specified RobotState is not marked as is_diff. The request to modify the object 'box' is not supported. Object is ignored.`
and also the stream of messages coming after. The box disappears. and the tutorial ends.

This however is not the same case in Kinetic. I am designing a pick and place application with UR5 arm, and still I am working in gazebo. Initially I thought it might be a problem with the moveit configuration for the arm. The planing for the arm works perfectly. I checked it via Rviz and also through my own node. the arm executes trajectories for the pose goals given. I also copied a part of the code of Move_group Interface tutorial [http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/move_group_interface_tutorial.html#adding-removing-objects-and-attaching-detaching-objects](url)

to check if the object gets connected to my gripper or not. that part of the code execute successfully without any errors. But there is a problem with **ROS API Planning Scene Tutorial**. 
I checked this tutorial for pr2 and for that the problem is the same. The object does not get detached and the execution does not end. 

`[ INFO] [1502275540.343626912]: Attached object 'box' to link 'r_wrist_roll_link'`

The process seems to hang over with the continuous streaming of messages which I think is a bug in between the transition call to remove. 
`planning_scene.world.collision_objects.clear();` 

your help is expected and appreciated to resolve this matter...!
OS Version: e.g. Ubuntu 16.04

BryceStevenWilley, 2017-09-07:
> [ERROR] [1502275537.653223449]: The specified RobotState is not marked as is_diff. The request to modify the object 'box' is not supported. Object is ignored.

The key is this error. Similar to [earlier](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/planning_scene_ros_api_tutorial.html#add-an-object-into-the-environment) in the tutorial, you have to set `planning_scene.is_diff =true`, you should also set `planning_scene.robot_state.is_diff = true`. I've made a PR #115 to fix this. 

> I have checked the same tutorial in indigo. it works in spite of
[ERROR] [1502275537.653223449]: The specified RobotState is not marked as is_diff. The request to modify the object 'box' is not supported. Object is ignored.
and also the stream of messages coming after. The box disappears. and the tutorial ends.

Can't speak to why this is the case however.

closed 2017-09-11 by v4hn

Hubert51, 2019-06-20:
How to set this flag in python interface?

PR #115 (2017-09-11): Added is 'is_diff = true' to fix detaching bug in moveit tutorial.
Should fix #98.