moveit/moveit_tutorials #107: RVIZ Visual tools GUI Does not appear

amirhm64, 2017-08-18:
Hi I am trying to use the movegroup interface tutorial on my own robot but I get this warning while running it 

      _Topic '/rviz_visual_tools' unable to connect to any subscribers within 0.5 sec. It is possible initially published visual messages will be lost._

and also on rviz side I cant see the rvizvisualtoolsgui to click next. On PR2 It runs perfectly.
I am running kinetic

amirhm64, 2017-08-21:
Had to add a new pannel in rviz

closed 2017-08-21 by amirhm64

WwYyFan, 2018-09-30:
@amirhm64  Hello, I could not found a pannel name "RvizVisualToolsGui" in rviz, how to fix it?

amirhm64, 2018-10-01:
> @amirhm64 Hello, I could not found a pannel name "RvizVisualToolsGui" in rviz, how to fix it?

you can go to panels>add new panel

stevewen, 2018-11-15:
@amirhm64 I have added the new pannel in rviz, but it  doesn't work, the warning still appear.

shd01x, 2019-07-08:
In case anyone's google leads here, you need to add a new rviz MarkerArray display, and make sure that the MarkerArray is subscribing to the topic that the markers are published on (that topic is a parameter to the MoveIt visual tools constructor in your C++ file).  If your setup is like mine, you might also need to add a delay of a few seconds before publishing so that Rviz has time to initialize.

SeanHRI, 2023-01-06:
You may first check whether there RvizVisualToolGui is. If not, I think you can solve this issue by typing "_sudo apt-get install ros-[distro]-rviz-visual-tools_" in your terminal.