moveit/moveit_tutorials #108: problem with catkin make

arielbis, 2017-08-22:
after I installed the pr2_moveit_config packege and git cloned the moveit_tutorials packege to my  src in the working space i had a problem with the catkin_make. thats the problem thath  it gave me:
[  0%] Built target roscpp_generate_messages_lisp
[  0%] [  0%] Built target tf2_msgs_generate_messages_cpp
Built target roscpp_generate_messages_cpp
[  0%] Built target diagnostic_msgs_generate_messages_py
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] [  0%] Built target std_msgs_generate_messages_lisp
Built target rosgraph_msgs_generate_messages_lisp
[  0%] [  0%] Built target rosgraph_msgs_generate_messages_py
Built target std_msgs_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] [  0%] [  0%] Built target sensor_msgs_generate_messages_py
Built target geometry_msgs_generate_messages_cpp
Built target diagnostic_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] [  0%] Built target sensor_msgs_generate_messages_lisp
Built target geometry_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_lisp
[  0%] [  0%] Built target trajectory_msgs_generate_messages_py
Built target nav_msgs_generate_messages_py
[  0%] [  0%] Built target actionlib_msgs_generate_messages_py
Built target diagnostic_msgs_generate_messages_cpp
[  0%] Built target actionlib_msgs_generate_messages_cpp
[  0%] [  0%] [  0%] Built target naoqi_bridge_msgs_generate_messages_lisp
Built target nav_msgs_generate_messages_lisp
Built target naoqi_bridge_msgs_generate_messages_py
[  0%] [  0%] Built target naoqi_bridge_msgs_generate_messages_cpp
Built target trajectory_msgs_generate_messages_lisp
[  0%] [  0%] Built target nav_msgs_generate_messages_cpp
Built target trajectory_msgs_generate_messages_cpp
[  0%] Built target actionlib_generate_messages_cpp
[  0%] [  0%] [  0%] Built target actionlib_generate_messages_lisp
Built target actionlib_generate_messages_py
Built target tf2_msgs_generate_messages_lisp
Scanning dependencies of target kinematic_model_tutorial
[  0%] Built target tf2_msgs_generate_messages_py
Scanning dependencies of target ros_api_tutorial
Scanning dependencies of target motion_planning_api_tutorial
Scanning dependencies of target move_group_interface_tutorial
[  1%] Building CXX object moveit_tutorials/doc/pr2_tutorials/kinematics/CMakeFiles/ros_api_tutorial.dir/src/ros_api_tutorial.cpp.o
[  3%] [  5%] Building CXX object moveit_tutorials/doc/pr2_tutorials/kinematics/CMakeFiles/kinematic_model_tutorial.dir/src/kinematic_model_tutorial.cpp.o
Building CXX object moveit_tutorials/doc/pr2_tutorials/planning/CMakeFiles/motion_planning_api_tutorial.dir/src/motion_planning_api_tutorial.cpp.o
[  6%] Building CXX object moveit_tutorials/doc/pr2_tutorials/planning/CMakeFiles/move_group_interface_tutorial.dir/src/move_group_interface_tutorial.cpp.o
/home/ariel/catkin_ws/src/moveit_tutorials/doc/pr2_tutorials/planning/src/move_group_interface_tutorial.cpp:37:62: fatal error: moveit/move_group_interface/move_group_interface.h: No such file or directory
 #include <moveit/move_group_interface/move_group_interface.h>
                                                              ^
compilation terminated.
make[2]: *** [moveit_tutorials/doc/pr2_tutorials/planning/CMakeFiles/move_group_interface_tutorial.dir/src/move_group_interface_tutorial.cpp.o] Error 1
make[1]: *** [moveit_tutorials/doc/pr2_tutorials/planning/CMakeFiles/move_group_interface_tutorial.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Linking CXX executable /home/ariel/catkin_ws/devel/lib/moveit_tutorials/kinematic_model_tutorial
[  6%] Built target kinematic_model_tutorial
Linking CXX executable /home/ariel/catkin_ws/devel/lib/moveit_tutorials/ros_api_tutorial
[  6%] Built target ros_api_tutorial
Linking CXX executable /home/ariel/catkin_ws/devel/lib/moveit_tutorials/motion_planning_api_tutorial
[  6%] Built target motion_planning_api_tutorial
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
 
what is missing that it cant compile?

srsidd, 2017-08-24:
Seems like you're missing the `move_group_interface` library? Have you installed `moveIt` fully?

jwhendy, 2017-08-24:
@srsidd beat me to it, but I was going to add that these issues are helped immensely by system details and some other hunting. Note the key error from your build:

```
fatal error: moveit/move_group_interface/move_group_interface.h: No such file or directory
#include <moveit/move_group_interface/move_group_interface.h>
```

Some handy things in troubleshooting posts:

```
# system
$ lsb_release -a
Description:	Ubuntu 16.04.3 LTS

# try to locate the file it says doesn't exist
$ locate move_group_interface.h
/opt/ros/kinetic/include/moveit/move_group_interface/move_group_interface.h

# I have the file, so if it helps, here's the package that owns it:
$ dpkg -S move_group_interface.h
ros-kinetic-moveit-ros-planning-interface: /opt/ros/kinetic/include/moveit/move_group_interface/move_group_interface.h
```

The distribution helps, as when I google `"move_group_interface.h"`, I see posts related to building the `kinetic-devel` branch on ROS indigo ([1](https://answers.ros.org/question/257250/moveit-build-error-move_group_interfaceh-file-not-found/) and [2](https://github.com/ros-planning/moveit_tutorials/issues/61)).

More likely, as suggested above, you don't have the right packages installed. It's in MoveIt!, as evidenced by the path (`moveit/move_group_interface.h`), so [check the install instructions](http://moveit.ros.org/install/) and post back with what you find.

Hope that helps!

arielbis, 2017-08-27:
thank you very much!
i got another question i tried to launch the naoqi_driver.launch file and
got the next messege:

... logging to
/home/peppercomp/.ros/log/73cf3b22-8b27-11e7-8111-d8cb8a065d13/roslaunch-peppercomp-ThinkCentre-M93p-1463.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://peppercomp-ThinkCentre-M93p:41325/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    naoqi_driver (naoqi_driver/naoqi_driver_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[naoqi_driver-1]: started with pid [1485]
[W] 1495 qimessaging.transportsocket: connect: Connection refused
terminate called after throwing an instance of 'qi::FutureUserException'
  what():  System error: Connection refused
================================================================================REQUIRED
process [naoqi_driver-1] has died!
process has died [pid 1485, exit code -6, cmd
/opt/ros/indigo/lib/naoqi_driver/naoqi_driver_node --qi-url=tcp://:9559
--roscore_ip=127.0.0.1 --network_interface=eth0 __name:=naoqi_driver
__log:=/home/peppercomp/.ros/log/73cf3b22-8b27-11e7-8111-d8cb8a065d13/naoqi_driver-1.log].
log file:
/home/peppercomp/.ros/log/73cf3b22-8b27-11e7-8111-d8cb8a065d13/naoqi_driver-1*.log
Initiating shutdown!
================================================================================
[naoqi_driver-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

do you und3erstand what the problem is?

jwhendy, 2017-08-27:
Few comments on customary practices with respect to troubleshooting issues:

- within a thread/issue, keep the discussion dedicated to the original issue(s) described
- that issue should be relevant to the place it's posted (`ros-planning` and `moveit` in this case)
- if the issue is resolved, click the "Comment and close" button to indicate that
- confirm what the resolution was for others who might find this post (I *assume* that "thank you very much!" means you don't have the error during `catkin make` anymore?)

For the new error, the second bullet would advise to take it to those more familiar (from a google, guessing [here](https://github.com/ros-naoqi/naoqi_driver)).

v4hn, 2017-09-11:
Closing. The discussion here went off-topic.

The original problem was that the requester tried to build `moveit_tutorials` in branch `kinetic-devel` with moveit branch `indigo-devel`.

closed 2017-09-11 by v4hn