moveit/moveit_tutorials #112: Pick and place not working

sweetsaikiran690, 2017-09-01:
Hi,
    I am very new to Moveit. I am following your tutorials. But I stucked at pick_place tutorial.
When I run the pick_and_place_tutorial.cpp file,
Table and objects are appearing in Moveit.
But after that it is showing error like

[ WARN] [1504262805.028268117]: Fail: ABORTED: No motion plan found. No execution attempted.
[ WARN] [1504262806.029454847]: Fail: ABORTED: Must specify group in motion plan request

Please help me to solve this issue.

Thanks
Sai kiran

tsouthard, 2018-05-10:
I am receiving the same error messages, have you figured out this problem? I am using the same cpp as you with the kuka youbot on ros indigo/ubuntu14.04.
I get the following warning when I launch the youbot_moveit_config demo.launch then run the pick place tutorial node
[ WARN] [1525979017.981283539]: All supplied grasps failed. Retrying last grasp in verbose mode.

the following error appears after running the pick place tutorial node as well
[ WARN] [1525979017.986390415]: Fail: ABORTED: No motion plan found. No execution attempted.
[ WARN] [1525979019.088840351]: Fail: ABORTED: Must specify group in motion plan request

Is there any more insight you can shed on this problem? is it something in the moveit assistant i need to change? or maybe the launch file or some controller file in the config?

davetcoleman, 2018-05-14:
This should be fixed in the new MoveIt! tutorials (not yet officially launched):
http://tutorials.picknik.ai

felixvd, 2018-07-23:
I just tested the kinetic pick and place tutorial locally, so I suppose this is solved. I did have issues with packages missing after the install instructions, but that's for another issue.

mlautman, 2018-07-23:
Thanks @felixvd!
@Ridhwanluthra  Can you and @felixvd address the issue of adding the necessary packages to the package.xml?

felixvd, 2018-07-24:
I guess it fits into this issue. When I set up an Ubuntu 16.04 machine like this:

```
apt-get update
apt-get install ros-kinetic-desktop-full
apt-get install ros-kinetic-moveit
cd ~\ws_moveit/src
git clone https://github.com/ros-planning/moveit_tutorials.git
git clone https://github.com/ros-planning/panda_moveit_config.git
rosdep install -y --from-paths . --ignore-src --rosdistro kinetic
```

Then I still need these to get it working:

```
git clone --single-branch -b kinetic-devel https://github.com/ros-planning/moveit_visual_tools.git
git clone https://github.com/PickNikRobotics/rviz_visual_tools.git
```

I only tested it in my docker container. Maybe someone could confirm.

felixvd, 2018-07-26:
Which problem? What version of ROS and Ubuntu are you using?

DebasmitaGhose, 2018-07-27:
I have opened a new issue #199. I am using ROS Kinetic and Ubuntu 16.04

mlautman, 2018-07-31:
@felixvd I don't believe that you should need those two additional lines to build the tutorials from scratch since moveit_visual_tools is a dependency in the package.xml. Can you confirm this?

felixvd, 2018-07-31:
I just did it again and it worked. Maybe I overlooked something last time, although I am pretty sure that no errors came up. My bad, all good.

mlautman, 2018-07-31:
I am closing this as I believe it has been resolved

closed 2018-07-31 by mlautman