moveit/moveit_tutorials #122: How to create a custom sampler?

qqfly, 2017-10-07:
As mentioned in [custom_constraint_samplers](https://github.com/ros-planning/moveit_tutorials/blob/a0f75a9df840ac3dfd0eec337cea104a4c111747/doc/custom_constraint_samplers.rst), we can create a custom sampler in MoveIt! But, I cannot find the hrp2jsk_moveit_constraint_sampler_plugin template.

BryceStevenWilley, 2018-11-03:
Partially correct: for the HRP-2, the closest things I can find are a [moveit config](https://github.com/start-jsk/rtmros_gazebo/tree/master/hrpsys_moveit_configs/hrp2jsk_moveit_config) for the HRP-2, and a [blog post](https://www.robotshop.com/community/blog/show/hrp2-jsk-is-pr2-best-friend) saying that the HRP-2 ran ROS in 2010, pretty early, along with the PR2.  It seems that when Willow was writing MoveIt docs, they probably had access to HRP2's code, but it's probably safe to say that code isn't available anymore. 

However, Dave Coleman seems to have his [own implementation](https://github.com/davetcoleman/moveit_humanoid_stability/blob/indigo-devel/include/moveit_humanoid_stability/humanoid_constraint_sampler.h) of a constraint sampler, possibly adapted from the JSK code.  @davetcoleman, would you be willing to link to your `HumanoidConstraintSampler` here?

davetcoleman, 2018-11-03:
I created that tutorial long ago and never shared my humanoid constraint sampler because the repo contained hrp2 urdf and meshes which are not open source.

qqfly, 2022-02-12:
It seems the docs have been updated, close this issue.

closed 2022-02-12 by qqfly