moveit/moveit_tutorials #128: Document how to use Gazebo with MoveIt!

davetcoleman, 2017-10-19:
We need a tutorial that discusses using ROS Control with Gazebo to plan in MoveIt!.

A really out of date version is [here](http://picknik.io/moveit_wiki/index.php?title=PR2/Gazebo/Quick_Start) but I recommend it not be followed because most of it is wrong.

felixvd, 2018-07-23:
That link is broken by now too. Maybe you mean [this](http://gazebosim.org/tutorials/?tut=ros_control).

cardboardcode, 2019-11-20:
Verified that [the link](http://gazebosim.org/tutorials/?tut=ros_control) does provide a tutorial that exemplifies the use of **ROS Control** with **Gazebo** to plan in **MoveIt**. 

## `WARNING`
However, do note that the code used in the tutorial is based in **ROS Kinetic** and therefore outdated due to the approaching **End-of-Life(EOL)** for **ROS Kinetic Kame Distribution**.  

@davetcoleman 
May we close this issue?

felixvd, 2019-11-20:
Considering [this new step](https://ros-planning.github.io/moveit_tutorials/doc/setup_assistant/setup_assistant_tutorial.html#step-9-gazebo-simulation) has been added to the moveit_setup_assistant tutorial in #184 which touches on Gazebo, I would say yes.

davetcoleman, 2019-11-20:
No, I think we could still really use a full featured tutorial on MoveIt/Gazebo integration...

cardboardcode, 2019-11-21:
@davetcoleman Noted. !smile!

Will start looking into it. However, due to my limited bandwidth for now, I am unable to set a definite deadline on when this issue can be properly addressed by.

Will update here once I have made some progress.

pringithub, 2020-04-30:
Please include one .. I'm doing the tutorial mentioned by @felixvd and can't get the visuals of the panda arm to show up .... I've been trying different things for hours

Kinetic 16.04
Gazebo9 (not 7)

also verified with colleague
Melodic 18.04
Gazebo9

JafarAbdi, 2020-04-30:
@pringithub Have a look at this [post](https://discourse.ros.org/t/moveit-gazebo-integration/10553/3), hopefully it will help you fix the issue you have right now

pringithub, 2020-05-01:
@JafarAbdi  I saw that post but simply wanted an out of the box working arm and didn't want to invest more time into debugging someone else's code. I ended up just pulling from the ur-gazebo repo which worked nicely https://github.com/ros-industrial/universal_robot.

pringithub, 2020-05-01:
It appears that the collision for the panda arm were there - turning on collision view showed the orange outline of the arm - however the visuals weren't loading and there were no debugging warnings/errors.

Moreover the ros_control was not working out-of-the-box, so a moveit newbie like me became quite frustrated

mlautman, 2020-05-07:
It seem like this is a request for additional work. Maybe you can put together a PR with a new tutorial for using MoveIt with gazebo?

davetcoleman, 2020-06-02:
I've just created a stub for this tutorial, would be great to have someone work on this for WMD 2020:
https://github.com/ros-planning/moveit_tutorials/pull/493