moveit/moveit_tutorials #154: How to "Allow External Comm"  in Joystick Control Teleoperation

oPiZiL, 2018-02-27:
In the [tutorial ](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/ros_visualization/joystick.html),

> In the Motion Planning plugin of Rviz, enable “Allow External Comm.” checkbox in the “Planning” tab. Enable the ‘Query Goal State’ robot display in the MoveIt! Motion Planning Plugins’s ‘Planning Request’ section.

I CANNOT find **Allow External Comm** checkbox.
My ros version is kinetic by the way.
Thanks, plz.

davetcoleman, 2018-03-26:
@mlautman is working on this tutorial so can comment

He11oworldd, 2024-11-01:
Hi, I have the same problem as yours, have you solved it?

akrbot, 2024-11-20:
in the motion planning in rviz ,
in planning Tab which is opened by default where u can plan and execute , also change planning groups etc
u can see Option section on the left side
there is External Comm check box.. check it and  u should be able to use joystic to set pose , plan and execute

> Hi, I have the same problem as yours, have you solved it?