moveit/moveit_tutorials #164: Moveit_Chomp motion planning with franka robot arm

shyamashi, 2018-04-25:
I am trying to use chomp motion planner with 7-DOF franka robot arm (https://frankaemika.github.io/docs/installation.html). Followed the tutorial for chomp interface in http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/chomp_interface_tutorial.html. The chomp motion planner is working with the fanuc robot but an error comes up with franka.
 Using ROS kinetic, Ubuntu 16.04. Built MoveIt from source.
[image]

[image]

@davetcoleman  @v4hn : Any help please. I am stuck in my project.

mlautman, 2018-04-27:
Good catch! This helped me find a bug in chomp where it was requesting the parent joint of a fixed base joint without checking for Null. 

I opened a PR to MoveIt that fixes the issue.

https://github.com/ros-planning/moveit/pull/870

shyamashi, 2018-04-30:
Thank you @mlautman for the reply. But unfortunately the same error " [move_group-4] process has died [pid 11986, exit code -11," repeats. 
I have replaced the src/moveit/moveit_planners/chomp/chomp_motion_planner/src/chomp_optimizer.cpp with the file given in https://github.com/PickNikRobotics/moveit/blob/6da46b9741dd32e77a215d4019ccd24c94dabbf3/moveit_planners/chomp/chomp_motion_planner/src/chomp_optimizer.cpp
Is there any other place I need to make a change? I'm still learning ROS.

mlautman, 2018-04-30:
It's hard to debug your issue unless you have a reproducible setup. I wouldn't overwrite the file but instead check out the branch from the picknikrobotics fork. Also, are you using the panda_moveit_config from picknikrobotics/panda_moveit_config or from the frankaemika/franka_ros/panda_moveit_config?

Maybe a full rebuild? 
catkin clean -y 
catkin build

shyamashi, 2018-04-30:
I was using frankaemika/franka_ros/panda_moveit_config.
This probably may sound stupid. But I have a 7-dof kuka robot arm model for which the chomp works perfectly where as the above error comes for 7-dof franka arm (did not load the gripper). So I used nautilus to compare the folders kuka_moveit_config and panda_moveit_config. The only major difference that I could find is that kuka generates urdf from xml whereas franka does the same from xacro (inside franka_description/robots). So I tried replacing the working kuka urdf with franka xacro and found that the error comes whereas viceversa appears to work properly. Very minor changes like some names of arms are different. I still could not pinpoint where exactly the problem is. If the above rings a bell, kindly let me know. 
If nothing works, then as you suggested a full rebuild seems to be the only option.
Thanks

shyamashi, 2018-05-01:
@mlautman : The panda_arm.xacro does not have a fixed joint which as per your first post should be corrected by adding those lines in the chomp_optimizer.cpp. But for me the error repeated. So I simply added a dummy world_link with no mass and inertia and fixed it with panda_link0 and it solved the error.

The chomp motion planner now works for Franka robot arm but some strange things happen. If I use the " Select Goal State -->random valid --Update" in Rviz, the chomp fails with the following terminal output
[image]

For chomp motion planner to be successful, I have to move the interactive marker at the end-effector with mouse and then click Plan -- execute in Rviz. Still there is an error (Found empty Joint State Message) but the output displays successfull execution of the trajectory and is as follows
[image]

I guess this is something specific to chomp algorithm as none of this happens with OMPL

mlautman, 2018-05-04:
I don't have the same issue with the `franka_ros/panda_moveit_config` package... can you provide a `.rosinstall` with the repos and branches in your workspace?

shyamashi, 2018-05-09:
Sorry for the delay.
The fixed joint problem is eliminated when I built moveit from source yesterday  (without adding any dummy world link)
The rosinstall file is 
 THIS IS AN AUTOGENERATED FILE, LAST GENERATED USING wstool ON 2018-05-08
- git:
    local-name: geometric_shapes
    uri: https://github.com/ros-planning/geometric_shapes.git
    version: kinetic-devel
- git:
    local-name: moveit
    uri: https://github.com/ros-planning/moveit.git
    version: kinetic-devel
- git:
    local-name: moveit_msgs
    uri: https://github.com/ros-planning/moveit_msgs.git
    version: jade-devel
- git:
    local-name: moveit_resources
    uri: https://github.com/ros-planning/moveit_resources.git
    version: master
- git:
    local-name: moveit_visual_tools
    uri: https://github.com/ros-planning/moveit_visual_tools.git
    version: kinetic-devel
- git:
    local-name: srdfdom
    uri: https://github.com/ros-planning/srdfdom.git
    version: kinetic-devel

The libfranka and franka_ros are from
https://github.com/frankaemika/libfranka
https://github.com/frankaemika/franka_ros

As I mentioned in my previous reply, When I click the 'plan and execute button' on the motion planning interface, it returns the error 'Start state is empty'. I had updated the start state through the interface 'Query-->Select start state--> update'. Also, another 'Warning: Fail:ABORTED Unknown event' happens when I try 'Query-->Select goal state--> update' and 'Plan'. 
[image]

SunYuansong, 2019-06-19:
Hello，
Is this problem solved?
I'm using abb irb1200+kinetic. I met the same problem when I want to use OMPL as a pre-processor for CHOMP, following the official MoveIt tutorial. But when I use the CHOMP alone, everything runs correctly. 
Any suggestion would be grateful

rhaschke, 2019-06-20:
This should have been fixed in #1473 for master branch.

closed 2019-06-20 by rhaschke