moveit/moveit_tutorials #168: bad PR / commit practice

rhaschke, 2018-05-16:
Obviously, since a while, you guys at PickNik only file PRs against the [PickNik fork](https://github.com/PickNikRobotics/moveit_tutorials) of this repo, but then fast-forward the ros-planning repo as well. This leads to commits in this repo that refer to misleading PRs, e.g. 85814dbb5f5ef93c752f8781eb90bd73cf42f626 referring to #55, which is completely unrelated! 
Please stop doing so and instead file PRs directly against the main repo at ros-planning.

The second drawback of this practice is, that you didn't notice that the `moveit_tutorials` package is broken in Travis since two months: https://travis-ci.org/ros-planning/moveit_tutorials/branches.
Please fix this asap. Since @davetcoleman added the tutorials package to the travis build of the main moveit repo in melodic, this is now failing too.

rhaschke, 2018-05-16:
Obviously the failing Travis is due to a change of the underlying Travis installation, now providing a more recent ruby version than the one we manually install. I'm experimenting with a fix in #167.

rhaschke, 2018-05-16:
Having fixed Travis' ruby issue, there pop up dozens of broken link warnings: https://travis-ci.org/ros-planning/moveit_tutorials/builds/379740334. Please fix them.

davetcoleman, 2018-05-16:
> instead of simply fast-forwarding this repo from your local one, a better approach might have been to explicitly merge.

I did a rebase-and-merge... I'm afraid I don't understand the difference enough but will try to research it more next time

rhaschke, 2018-05-16:
While a rebase tries to maintain a linear commit history, an explicit merge commit allows to recognize that a bunch of commits was merged in from somewhere else and it also indicates from where.

closed 2018-05-28 by v4hn