moveit/moveit_tutorials #174: incorrect link in installation page

jash101, 2018-05-25:
The link ( http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/ros_visualization/visualization_tutorial.html ) on the installation page under quickstart section is incorrect. Please correct it.

mlautman, 2018-05-29:
Hey @jash101 We appreciate the feedback but I don't see a broken link. Is this still broken? The buildfarm runs every few days.

If the link is still broken, please post a link with more exact instructions to find the hypertext on the tutorial page. 

Also, in addition to opening an issue I would encourage you to open a PR in the future. I can mentor you in how to do this if you run into blockers :)

davetcoleman, 2018-05-30:
@jash101 is referring to the bottom of:
http://moveit.ros.org/install/

Thanks for catching this! Do you mind making a PR to fix it? See:
https://github.com/ros-planning/moveit.ros.org/blob/gh-pages/install/index.markdown

jash101, 2018-05-31:
@davetcoleman sure I'll do that

jash101, 2018-05-31:
@davetcoleman I think it's already been fixed. Sorry for the late reply

davetcoleman, 2018-05-31:
Great. In the future be sure to close an issue if you think its resolved :)

closed 2018-05-31 by davetcoleman