moveit/moveit_tutorials #175: A PickPlace tutorial

v4hn, 2018-05-28:
The [MoveIt task constructor](https://github.com/moveit/moveit_task_constructor) is meant to replace the pick place pipeline eventually, but the current pipeline should *still* be documented in a tutorial.
The replacement might even use the same interfaces as a drop-in replacement.

The only documentation we have for this at the moment is a single C++ file without comments that works with the PR2... which recently got replaced by panda throughout this repository..
This is not useful, especially because it is non-trivial to set working values for all parameters.

Ridhwanluthra, 2018-06-04:
The approach given in the file seems to be a hard-coded implementation. is that correct?
if so, maybe [moveit_grasps](https://github.com/davetcoleman/moveit_grasps) can be used for designing pick place tutorial?

Ridhwanluthra, 2018-06-05:
I believe to build this tutorial we would need to setup gazebo as without that it's probably nothing different than the basic move group interface tutorial.

Ridhwanluthra, 2018-06-05:
@mlautman @feroze

mlautman, 2018-06-06:
I think Gazebo integration would be a great incredible tutorial on it's own. Adding gazebo integration to another tutorial would add complexity that deviates from the model of the tutorials which is, teach the user one thing at a time.

davetcoleman, 2018-06-06:
We do have a GSoC student working on Gazebo integration fyi.

Ridhwanluthra, 2018-06-19:
[PR](https://github.com/ros-planning/moveit_tutorials/pull/182)

DebasmitaGhose, 2018-07-24:
I have been trying to use this tutorial to implement a pick and place task for a custom robot. Can you please explain how are the values for the following variables assigned?

  grasps[0].grasp_pose.pose.position.x = 0.415;
  grasps[0].grasp_pose.pose.position.y = 0;
  grasps[0].grasp_pose.pose.position.z = 0.5;

I am attempting grasp by using two hands and I have obtained the position of the end effectors in the frame of the robot. Which of these values should I be using?
rosrun tf tf_echo base_footprint lh_palm
At time 1532437665.651
- Translation: [0.727, 0.346, 1.017]
- Rotation: in Quaternion [-0.103, 0.640, -0.253, 0.718]
            in RPY (radian) [-1.247, 1.050, -1.466]
            in RPY (degree) [-71.426, 60.170, -83.993]
 
rosrun tf tf_echo base_footprint rh_palm
At time 1532437942.251
- Translation: [0.729, -0.285, 1.004]
- Rotation: in Quaternion [-0.175, 0.612, -0.074, 0.768]
            in RPY (radian) [-1.085, 1.152, -0.939]
            in RPY (degree) [-62.179, 66.005, -53.797]

Ridhwanluthra, 2018-07-29:
Right above these lines in the comments, it is mentioned how these values are set. [check here](https://github.com/ros-planning/moveit_tutorials/blob/kinetic-devel/doc/pick_place/src/pick_place_tutorial.cpp#L84)
I don't think there is a direct function that you can call for a 2 hand grasp. maybe wait for someone else to comment on this.

Ridhwanluthra, 2018-08-02:
@v4hn we can close this issue

mlautman, 2018-08-02:
Thanks for your work on the [Pick and Place Tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pick_place/pick_place_tutorial.html) @Ridhwanluthra!

closed 2018-08-02 by mlautman