moveit/moveit_tutorials #176: process died in robot_model_and_robot_state_tutorial

evancsj, 2018-05-31:
I tried to run robot_model_and_robot_state_tutorial by typing 
```
roslaunch moveit_tutorials robot_model_and_robot_state_tutorial.launch 
```
But I got error as
```
... logging to /home/csj/.ros/log/c6857aee-64a6-11e8-b694-7c7a91ff74c2/roslaunch-csj-ThinkPad-T440p-31579.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://csj-ThinkPad-T440p:33045/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /robot_description_kinematics/panda_arm/kinematics_solver: kdl_kinematics_pl...
 * /robot_description_kinematics/panda_arm/kinematics_solver_attempts: 3
 * /robot_description_kinematics/panda_arm/kinematics_solver_search_resolution: 0.005
 * /robot_description_kinematics/panda_arm/kinematics_solver_timeout: 0.005
 * /robot_description_planning/joint_limits/panda_finger_joint1/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/panda_finger_joint1/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_finger_joint1/max_acceleration: 0
 * /robot_description_planning/joint_limits/panda_finger_joint1/max_velocity: 0.3
 * /robot_description_planning/joint_limits/panda_finger_joint2/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/panda_finger_joint2/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_finger_joint2/max_acceleration: 0
 * /robot_description_planning/joint_limits/panda_finger_joint2/max_velocity: 0.3
 * /robot_description_planning/joint_limits/panda_joint1/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/panda_joint1/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_joint1/max_acceleration: 0
 * /robot_description_planning/joint_limits/panda_joint1/max_velocity: 2.5
 * /robot_description_planning/joint_limits/panda_joint2/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/panda_joint2/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_joint2/max_acceleration: 0
 * /robot_description_planning/joint_limits/panda_joint2/max_velocity: 2.5
 * /robot_description_planning/joint_limits/panda_joint3/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/panda_joint3/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_joint3/max_acceleration: 0
 * /robot_description_planning/joint_limits/panda_joint3/max_velocity: 2.5
 * /robot_description_planning/joint_limits/panda_joint4/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/panda_joint4/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_joint4/max_acceleration: 0
 * /robot_description_planning/joint_limits/panda_joint4/max_velocity: 2.5
 * /robot_description_planning/joint_limits/panda_joint5/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/panda_joint5/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_joint5/max_acceleration: 0
 * /robot_description_planning/joint_limits/panda_joint5/max_velocity: 3
 * /robot_description_planning/joint_limits/panda_joint6/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/panda_joint6/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_joint6/max_acceleration: 0
 * /robot_description_planning/joint_limits/panda_joint6/max_velocity: 3
 * /robot_description_planning/joint_limits/panda_joint7/has_acceleration_limits: False
 * /robot_description_planning/joint_limits/panda_joint7/has_velocity_limits: True
 * /robot_description_planning/joint_limits/panda_joint7/max_acceleration: 0
 * /robot_description_planning/joint_limits/panda_joint7/max_velocity: 3
 * /robot_description_semantic: <?xml version="1....
 * /robot_model_and_robot_state_tutorial/panda_arm/kinematics_solver: kdl_kinematics_pl...
 * /robot_model_and_robot_state_tutorial/panda_arm/kinematics_solver_attempts: 3
 * /robot_model_and_robot_state_tutorial/panda_arm/kinematics_solver_search_resolution: 0.005
 * /robot_model_and_robot_state_tutorial/panda_arm/kinematics_solver_timeout: 0.005
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    robot_model_and_robot_state_tutorial (moveit_tutorials/robot_model_and_robot_state_tutorial)

auto-starting new master
process[master]: started with pid [31592]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to c6857aee-64a6-11e8-b694-7c7a91ff74c2
process[rosout-1]: started with pid [31605]
started core service [/rosout]
process[robot_model_and_robot_state_tutorial-2]: started with pid [31622]
[ INFO] [1527752810.009658810]: Loading robot model 'panda'...
[robot_model_and_robot_state_tutorial-2] process has died [pid 31622, exit code -11, cmd /home/csj/catkin_ws/devel/lib/moveit_tutorials/robot_model_and_robot_state_tutorial __name:=robot_model_and_robot_state_tutorial __log:=/home/csj/.ros/log/c6857aee-64a6-11e8-b694-7c7a91ff74c2/robot_model_and_robot_state_tutorial-2.log].
log file: /home/csj/.ros/log/c6857aee-64a6-11e8-b694-7c7a91ff74c2/robot_model_and_robot_state_tutorial-2*.log
```
Anyone know what's wrong?

mlautman, 2018-07-12:
You're going to need to give us more information about your setup for us to be able to help you. That tutorial works consistently on all of my machines.

QNjeason, 2020-05-05:
I have this problem, too.  In melodic,18.04