moveit/moveit_tutorials #181: undefined reference to `moveit_visual_tools::MoveItVisualTools::MoveItVisualTools

HankerSia, 2018-06-15:
hello, when i tried to run the demo,I got the following errors:
basicControl.cpp:205: undefined reference to `moveit_visual_tools::MoveItVisualTools::MoveItVisualTools(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<moveit::core::RobotModel const>)'
basicControl.cpp:206: undefined reference to `rviz_visual_tools::RvizVisualTools::deleteAllMarkers()'

the code snippet is as follow:
namespace rvt = rviz_visual_tools;
moveit_visual_tools::MoveItVisualTools visual_tools("panda_link0");
visual_tools.deleteAllMarkers();

I use ubuntu16.04 and kinetic, Is this a bug?

v4hn, 2018-06-18:
This is a linker error and has nothing to do with the code.
Apparently you copied some code into your own package and forgot to find_package `moveit_visual_tools` (and possibly `rviz_visual_tools` if you use that too) or didn't link it to your cmake target.

HankerSia, 2018-06-19:
Ok, get it. I forgot that. Thank you!

closed 2018-06-25 by v4hn