moveit/moveit_tutorials #183: Problem defining Robot Pose in moveit setup assistant

parthvasoya, 2018-06-20:
Hello,
i have recently starting working with ROS and i am currently stuck because of a very wired problem.
i am not able to define Robot Pose in moveit setup assistant as the joints are not displayed on right side as it should be. please refer the snap below:
[image]
Appreciate any help!

mlautman, 2018-06-21:
The setup assistant has been getting a lot of changes recently. Can you tell us more about your setup?
- Are you building moveit from source or debian?
- What UR3 package are you using?
- Do you have the same behavior if you use the moveit_setup_assistant with the panda arm in  franka_decscription?
- Why are you recreating the UR3 package when you can get one from [here](https://github.com/ros-industrial/universal_robot)?

gavanderhoorn, 2018-06-30:
@mlautman: I don't believe that is a UR3. They don't bend at the elbow like the robot does in the screenshot.

Also: should this not be reported over at `ros-planning/moveit`?

mlautman, 2018-07-12:
Looks like the new E-series https://www.roboticsbusinessreview.com/manufacturing/universal-robots-launches-e-series-cobot/

gavanderhoorn, 2018-07-14:
@mlautman: you're right, I hadn't considered those.

In that case it would be clear why @parthvasoya is creating his own MoveIt configuration: `ur_description` does not support any of the e-series robots right now.

mlautman, 2018-09-19:
@parthvasoya could be an issue with your urdf. Can you share the files you are passing into the setup assistant?