moveit/moveit_tutorials #186: IKFast tutorial: use Docker instead of from-source OpenRAVE

gavanderhoorn, 2018-06-30:
The standard (and my standard) response would be "ok, fix it yourself", but I won't have time for this in the coming two months, so wanted to at least track this here in order to give other people a chance to perhaps pick this up.

The current IKFast tutorial seems too convoluted right now. Installing from source and the whole OpenRAVE installation should not be necessary.

The [personalrobotics/ros-openrave](https://hub.docker.com/r/personalrobotics/ros-openrave/) Docker image contains all the necessary pieces and is only about 500MB. Using it avoids 'polluting' the user's machine with a lot of dependencies OpenRAVE needs when building it from sources, and the image also works on other OS than Ubuntu.

gavanderhoorn, 2018-06-30:
Related: #187.

felixvd, 2018-07-26:
Are you proposing to install it in its own container and only use its output afterwards? Can you call it from outside the container or do users need to enter it?

I like the idea but I don't know if teaching beginners about docker and shells might not turn out just as long.

gavanderhoorn, 2018-07-27:
No need to install anything or teach anyone about Docker. See #187 for some example workflows (that can be streamlined quite a bit).

mwamp, 2018-12-09:
I strongly support adding this to the tutorial. Here is a piece of my experience with this as a beginner.
I am running Ubuntu bionic, there are no binaries for ros-openrave in the melodic repo. So I tried compiling openrave from source. Of course I ran into trouble since some dependencies of openrave needed outdated versions of boostLibs. Then I tried a quickfix by adding some boost headers from older boost version to get things to compile. Things got messy. You have to edit a few makefile you didn't write, find a few fixes on the internet for stange but reference boost bugs etc... Of course later I would also need the binaries, which I installed.
Ultimately I found an old ssd with ubuntu 16.04 and relevant boostLibs on which I could install OpenRave. And I found myself switch between my system and an external boot for a few days.

I don't mean to complain or share my lifestory. I want to stress this is the kind of things that would tend to happen to people new to this tool and who don't necessarily post a SOS question in a situation where you just have to get your own dependencies right and know it will work afterwards. 

In comparison installing docker and running openrave inside it took me less than a half hour today (I don't know or use docker). I think the tutorial definitely deserves at the very least a pointer to this option. What's more you probably won't use openrave everyday, that makes a container a very good fit in my opinion.

felixvd, 2018-12-17:
Since you just went through the process and it's still fresh in your memory, how about writing down the steps and adding them to the tutorial?