moveit/moveit_tutorials #188: panda does not move in motion_planning_api_tutorial

hankyang94, 2018-07-02:
I am following the [motion_planning_api_tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/motion_planning_api/motion_planning_api_tutorial.html), my system is ROS kinetic and Ubuntu 16.04.
There are two issues with this tutorial, which I am surprised that I could not find any similar issues and solutions to it.

1: console outputs error:
`[ INFO] [1530492597.516038871]: Failed to call service get_planning_scene, have you launched move_group? at /tmp/binarydeb/ros-kinetic-moveit-ros-planning-0.9.12/planning_scene_monitor/src/planning_scene_monitor.cpp:485
`
I have seen similar posts about this issue, but there seems to be no solution yet. And some people say this should just be a warning instead of an error. Is this true?

2: the panda arm does not move even though solution is found.
The console outputs:
`[ERROR] [1530492689.634402542]: Found empty JointState message
[ INFO] [1530492689.634574456]: The timeout for planning must be positive (0.000000 specified). Assuming one second instead.
[ WARN] [1530492689.634755603]: It looks like the planning volume was not specified.
[ WARN] [1530492689.635149109]: Returning dirty link transforms
[ WARN] [1530492689.635364041]: Returning dirty link transforms
[ WARN] [1530492689.635715756]: Returning dirty link transforms
[ WARN] [1530492689.635739751]: Returning dirty link transforms
[New Thread 0x7fffd11c7700 (LWP 3044)]
[ INFO] [1530492689.636269834]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1530492689.639705328]: RRTConnect: Starting planning with 1 states already in datastructure
[Thread 0x7fffd11c7700 (LWP 3044) exited]
[ INFO] [1530492689.681683929]: RRTConnect: Created 6 states (2 start + 4 goal)
[ INFO] [1530492689.681715274]: Solution found in 0.043579 seconds
[ INFO] [1530492689.690989722]: SimpleSetup: Path simplification took 0.009124 seconds and changed from 5 to 2 states
[ INFO] [1530492689.694297461]: Visualizing the trajectory
`
But I cannot see the arm move at all during the process. 

Could you help me identify what is wrong?

closed 2018-07-12 by mlautman

keep9oing, 2018-11-19:
same but change topic in rviz is not help

pranavb104, 2018-11-19:
I have the same issue...changing trajectory topic does not work....

HDallali, 2019-01-02:
I have the same issue and the trajectory topic is already set to /move_group/display_planned_path