moveit/moveit_tutorials #189: Cannot launch motion_planning_api_tutorial: error while loading shared libraries: libmoveit_planning_scene_monitor.so.0.9.12

limwenyao, 2018-07-06:
I am using the motion_planning_api_tutorial, on ROS kinetic and Ubuntu 16.04. I am unable to move the panda arm within Rviz GUI after launching the tutorial. It seems that error loading the `libmoveit_planning_scene_monitor.so.0.9.12` has caused the entire thing to fail.

```
auto-starting new master
process[master]: started with pid [25625]
ROS_MASTER_URI=http://10.12.161.36:11311

setting /run_id to 7e85798c-8121-11e8-b3f0-84ef18323a0c
process[rosout-1]: started with pid [25638]
started core service [/rosout]
process[virtual_joint_broadcaster_0-2]: started with pid [25655]
process[joint_state_publisher-3]: started with pid [25656]
process[robot_state_publisher-4]: started with pid [25664]
process[rviz_limwenyao_ubuntu_25612_9090462320018716399-5]: started with pid [25668]
process[motion_planning_api_tutorial-6]: started with pid [25680]
/home/limwenyao/catkin_ws/devel/lib/moveit_tutorials/motion_planning_api_tutorial: error while loading shared libraries: libmoveit_planning_scene_monitor.so.0.9.12: cannot open shared object file: No such file or directory
[motion_planning_api_tutorial-6] process has died [pid 25680, exit code 127, cmd /home/limwenyao/catkin_ws/devel/lib/moveit_tutorials/motion_planning_api_tutorial __name:=motion_planning_api_tutorial __log:=/home/limwenyao/.ros/log/7e85798c-8121-11e8-b3f0-84ef18323a0c/motion_planning_api_tutorial-6.log].
log file: /home/limwenyao/.ros/log/7e85798c-8121-11e8-b3f0-84ef18323a0c/motion_planning_api_tutorial-6*.log
[ INFO] [1530884149.969666302]: rviz version 1.12.15
[ INFO] [1530884149.969704414]: compiled against Qt version 5.5.1
[ INFO] [1530884149.969716809]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1530884150.078094313]: Stereo is NOT SUPPORTED
[ INFO] [1530884150.078191496]: OpenGl version: 3 (GLSL 1.3).
[ INFO] [1530884153.436142651]: Loading robot model 'panda'...
[ INFO] [1530884153.479770349]: Loading robot model 'panda'...
[ INFO] [1530884153.521687810]: Starting scene monitor
[ INFO] [1530884153.523959551]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1530884153.524699594]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ INFO] [1530884158.534360618]: Failed to call service get_planning_scene, have you launched move_group? at /tmp/binarydeb/ros-kinetic-moveit-ros-planning-0.9.11/planning_scene_monitor/src/planning_scene_monitor.cpp:486
[ INFO] [1530884159.015656349]: Constructing new MoveGroup connection for group 'panda_arm' in namespace ''
[ERROR] [1530884189.032431009]: Unable to connect to move_group action server 'move_group' within allotted time (30s)
[ WARN] [1530884189.041242916]: Interactive marker 'EE:goal_panda_link8' contains unnormalized quaternions. This warning will only be output once but may be true for others; enable DEBUG messages for ros.rviz.quaternions to see more details.
```

limwenyao, 2018-07-10:
@mlautman I did as you said, and I have gotten back the same error. I noticed when i build my workspace, the following warnings appeared.
```
Warnings   << moveit_tutorials:make /home/limwenyao/catkin_ws/logs/moveit_tutorials/build.make.000.log
/usr/bin/ld: warning: libmoveit_planning_scene_monitor.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_model.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_transforms.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_state.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_trajectory.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_collision_detection.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_planning_scene.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_planning_scene_monitor.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_model.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_transforms.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_state.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_trajectory.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_collision_detection.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_planning_scene.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_planning_scene_monitor.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_model.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_transforms.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_state.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_trajectory.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_collision_detection.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_planning_scene.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_planning_scene_monitor.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_model.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_transforms.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_state.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_trajectory.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_collision_detection.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_planning_scene.so.0.9.12, needed by /opt/ros/kinetic/lib/libmoveit_visual_tools.so, not found (try using -rpath or -rpath-link)
cd /home/limwenyao/catkin_ws/build/moveit_tutorials; catkin build --get-env moveit_tutorials | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Finished  <<< moveit_tutorials                      [ 37.6 seconds ]           
[build] Summary: All 18 packages succeeded!                                    
[build]   Ignored:   2 packages were skipped or are blacklisted.               
[build]   Warnings:  2 packages succeeded with warnings.                       
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 41.9 seconds total.                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.

```

mlautman, 2018-07-11:
I would try running 
`sudo apt update`
`sudo apt upgrade`

mlautman, 2018-07-12:
I'm closing this because it really isn't an issue with the tutorials but let me know if this doesn't work and we can keep debugging

closed 2018-07-12 by mlautman