moveit/moveit_tutorials #199: Pick and Place error

DebasmitaGhose, 2018-07-26:
I am trying to reproduce the pick and place code for my robot. I am getting the following error:

`[ INFO] [1532611370.581613456]: Ready to take commands for planning group robot_without_rhand_fingers.`
`[ WARN] [1532611371.660671687]: Fail: ABORTED: Must specify group in motion plan request`
`[ WARN] [1532611372.760226386]: Fail: ABORTED: Must specify group in motion plan request`

I am using ROS kinetic on Ubuntu 16.04. 
It would be great if someone could provide a solution for this error.

felixvd, 2018-07-26:
Please post your srdf file.

DebasmitaGhose, 2018-07-27:
This is the link to the srdf file
https://pastebin.com/mgAeWeXs

DebasmitaGhose, 2018-07-27:
And these are the more detailed logs of the error:
https://pastebin.com/D9vUKfVD

DebasmitaGhose, 2018-07-30:
Thank you for explaining this. 
I made those changes and I still get the same errors and I dont understand why. Here is a link to my code for pick and place
https://pastebin.com/tsbdpXGY
And my updated srdf file
https://pastebin.com/4hRSiqdV
It would be great if you could check and tell me where I am going wrong

felixvd, 2018-07-30:
1. openGripper needs to define the joints of the end effector instead of the robot. I also don't know what joints are included in "lh_arm" and "rh_arm" because I don't know your robot. The robot_without_rhand_fingers definition looks wild, it sounds like that group includes the whole robot, but you define that group as an end effector in your srdf. 

I think you should clean this up some more on your own and take a look at robot definitions that are working, copy what they are doing, and then branch out slowly instead of doing everything at once and hope it works. See [this two-armed HIRO](https://github.com/tork-a/rtmros_nextage/blob/indigo-devel/nextage_moveit_config/config/NextageOpen.srdf) or the [PR2](https://github.com/davetcoleman/pr2_moveit_config/blob/master/config/pr2.srdf).

2. Please make this a question on [ROS answers](https://answers.ros.org/questions/) instead and read [this](https://stackoverflow.com/help/how-to-ask) so you can post with a more helpful title that people can find later on. Github isn't really the place for troubleshooting of this sort.

v4hn, 2018-08-20:
> Github isn't really the place for troubleshooting of this sort.

:+1: 

Closing

closed 2018-08-20 by v4hn