moveit/moveit_tutorials #216: Planning Scene object attaching troubles

juliangaal, 2018-08-16:
I want to add grip and object from a table, and drop it off at at different position with the jac2 robotic arm.

### What works
Add object to planning scene
Attaching object to arm (box turns violet in rviz)

### What fails
I don't control the grippers with MoveIt!, but with the jaco drivers. Everything else is controlled with MoveIt! However, attaching the object should stop MoveIt! from checking for collisions around the hand, no? My pose planning happens around "jaco_link_hand", the hand joint, as seen in the picture below.

[image]

Now, when i close the grippers however, I get this error in 90% of all cases 
```bash
[ INFO] [/move_group]: Found a contact between 'jaco_link_finger_tip_3' (type 'Robot link') and 'cup' (type 'Robot attached'), which constitutes a collision. Contact information is not stored.
[ INFO] [/move_group]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [/move_group]: Start state appears to be in collision with respect to group arm
[ WARN] [/move_group]: Unable to find a valid state nearby the start state (using jiggle fraction of 0.050000 and 100 sampling attempts). Passing the original planning request to the planner.
```
jaco_link_finger_tip can be seen int the picture above, it's the last link in the fingers when looking from the top.

When I force my arm not to grip, everything works fine (except gripping the object, of course). I only sometimes get a collision with my tabletop, which is a simple collision object in MoveIt!.

So, is the only option I have to integrate the gripper into my moveit config? But how could i tell moveit to ignore the finger tips when gripping, as well?

Also, is there a way to tell moveit to ignore the bottom collision with my table, as it sometimes fails with

```bash
[ INFO] [/move_group]: Found a contact between 'table' (type 'Object') and 'cup' (type 'Robot attached'), which constitutes a collision. Contact information is not stored.
``` 

Thanks for your help

mlautman, 2018-08-20:
This isn't an issue with the tutorials. Please post questions like this to [https://answers.ros.org](https://answers.ros.org)

juliangaal, 2018-08-21:
did. Thanks

closed 2018-08-21 by juliangaal