moveit/moveit_tutorials #227: error: 'tf' has not been declared

Landanjs, 2018-09-25:
System: Ubuntu 18.04
ROS distro: Melodic

Following the steps on the MoveIt! getting started page https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html. I received the error "error: 'tf' has not been declared" in the pick_place_tutorial.cpp file.

 Following the advice at https://answers.ros.org/question/301306/error-tf-has-not-been-declared/ I was able to fix the error by adding "#include <tf/transform_broadcaster.h>" to pick_place_tutorial.cpp.

Please fix this for future users.

davetcoleman, 2018-10-13:
This indicates its time to fork the tutorials for Melodic:
https://github.com/ros-planning/moveit_tutorials/issues/230

Also this TF issue is related to:
https://github.com/ros-planning/moveit/issues/925

IanTheEngineer, 2018-10-15:
Seems I missed converting the tutorial files during the `tf2` upgrade. The solution linked by @Landanjs  is a workaround, but the permanent fix was to update moveit_tutorials to `tf2`.

> Also this TF issue is related to:
ros-planning/moveit#925

@davetcoleman This issue outlined in this ticket is a straightforward port to tf2 of the tutorials, and can be solved independent of #925.

closed 2018-10-16 by mlautman

davetcoleman, 2018-10-17:
> was to update moveit_tutorials to tf2.

@IanTheEngineer doesn't that break the Kinetic tutorials, which are currently shared by Melodic?

IanTheEngineer, 2018-10-17:
@davetcoleman I don't believe so, other than explicitly adding tf2 dependencies. These updates just use tf2-related packages which were released way back in Indigo. Most of the tf2 changes to MoveIt itself were similarly backwards compatible, except the API breaking changes. Nothing in the tutorials uses those updated API's. The only reason there was an error at all was that the tutorials relied on the fact that `tf1` would be included implicitly by depending on MoveIt.

IanTheEngineer, 2018-10-17:
Follow-on thought: the tutorials are only using `tf` for type <-> message conversions and static world transforms in the launch files. All of that was available in Kinetic. I built and tested my PR in both Kinetic and Melodic via docker container fwiw.