moveit/moveit_tutorials #230: Branch Tutorials for Meldoic

davetcoleman, 2018-10-13:
From this [issue](https://github.com/ros-planning/moveit_tutorials/issues/227#issuecomment-426743711) its become clear its time to branch the moveit tutorials for ROS Melodic. Any objections? I'm currently updating the README to explain the versioning process.

IanTheEngineer, 2018-10-16:
Issue #227 likely won't cause a branch for the tutorials, but it likely makes sense to have `melodic` keywords in the tutorials. Does that require a branch?

davetcoleman, 2018-10-17:
Yes it requires a branch and it will start pushing people to using melodic, which it feels like is the right time assuming your TF PR goes through :)

davetcoleman, 2018-10-26:
Yes we forgot to switch the github pages deploy script to use melodic, but @nbbrooks fixed it yesterday and I just merged it seconds ago:
https://github.com/ros-planning/moveit_tutorials/pull/244

davetcoleman, 2018-10-26:
The ros build farm won't have a melodic version available until @130s releases it via bloom.

closed 2018-10-26 by davetcoleman