moveit/moveit_tutorials #235: Workaround for missing franka_description package in Melodic

davetcoleman, 2018-10-18:
We need to document how to git clone the franka_ros repo on Melodic and possibly blacklist unnecessary packages (most of them).

See also
https://answers.ros.org/question/303278/moveit-tutorials-on-ros-melodic/
https://github.com/frankaemika/franka_ros/issues/34#issuecomment-430903870

mohamedghita, 2018-11-05:
My workaround on Ubuntu 18.04 & ROS Melodic is just git franka_description pkg from franka_ros repo.
Also, I needed to clone moveit_visual_tools into my catkin workspace after receiving error "E: Unable to locate package ros-melodic-moveit-visual-tools"

`git clone https://github.com/frankaemika/franka_ros.git`
`cp -a franka_ros/franka_description/ . && rm -rf franka_ros`

`git clone https://github.com/ros-planning/moveit_visual_tools.git`

I suggest adding the above lines in the documentation instead of the one line that clones ros_franka repo.

davetcoleman, 2018-11-09:
I believe this is fixed now as its been released to melodic.

Related: https://github.com/ros-planning/moveit/pull/1098

closed 2018-11-09 by rhaschke

Napoleon-BlownApart, 2018-11-14:
Following the [MoveIt Getting Started instructions](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html), when performing the step

    rosdep install -y --from-paths . --ignore-src --rosdistro melodic

I got an apt error stating:

    executing command [sudo -H apt-get install -y ros-melodic-moveit-visual-tools]
    Reading package lists... Done
    Building dependency tree       
    Reading state information... Done
    E: Unable to locate package ros-melodic-moveit-visual-tools
    ERROR: the following rosdeps failed to install
      apt: command [sudo -H apt-get install -y ros-melodic-moveit-visual-tools] failed

This is a fresh U18.04 guest installed today using VirtualBox on my host MacBook Pro (MacOS).  I've tried the workaround instructions above and will see how I go.

rhaschke, 2018-11-14:
@davetcoleman Binary release of moveit_visual_tools is broken since several weeks: http://build.ros.org/view/Mbin_uB64/job/Mbin_uB64__moveit_visual_tools__ubuntu_bionic_amd64__binary/171/console

Napoleon-BlownApart, 2018-11-14:
@rhaschke : Thanks, and I can see that the build fails.

So, I used the Debian package instead (which goes against the advice of mixing packages.ros with debian packages):

    sudo apt-get install ros-melodic-rviz-visual-tools

I then continued with the tutorial's _Build your Catkin Workspace_ section and the build worked. I now have RViz running with a Panda robot.

davetcoleman, 2018-11-14:
> Why is it released from PickNik's fork??

I'm not sure it was ever switched from there? Or I just got confused at some point. Fixing now, tracking progress here:
https://github.com/ros-planning/moveit_visual_tools/issues/38