moveit/moveit_tutorials #256: problem launching motion_planning_api_tutorial

keep9oing, 2018-11-13:
I'm doing tutorial of moveit! 
In section [motion planning api] 
I tried to run command

roslaunch moveit_tutorials motion_planning_api_tutorial.launch

But the error came out.
========================================================================
[ INFO] [1542094534.982780604]: Loading robot model 'panda'...
[ INFO] [1542094535.034559684]: rviz version 1.12.16
[ INFO] [1542094535.034772607]: compiled against Qt version 5.5.1
[ INFO] [1542094535.034821834]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1542094535.074079088]: Loading robot model 'panda'...
[ INFO] [1542094535.162254147]: Initializing OMPL interface using ROS parameters
[ INFO] [1542094535.182060369]: Stereo is NOT SUPPORTED
[ INFO] [1542094535.182157262]: OpenGl version: 3 (GLSL 1.3).
[ INFO] [1542094535.247378432]: Using planning interface 'OMPL'
[ INFO] [1542094535.262996856]: RemoteControl Ready.
[ INFO] [1542094535.466988253]: Loading robot model 'panda'...
[ERROR] [1542094535.481714714]: Ignoring transform for child_frame_id "100" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 0.000000 0.000000 0.000000)
[ INFO] [1542094535.939800520]: Loading robot model 'panda'...
[ INFO] [1542094536.002095107]: Loading robot model 'panda'...
[ INFO] [1542094536.056537778]: Starting scene monitor
[ INFO] [1542094536.059636542]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1542094536.061016046]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ INFO] [1542094541.067191576]: Failed to call service get_planning_scene, have you launched move_group? at /tmp/binarydeb/ros-kinetic-moveit-ros-planning-0.9.15/planning_scene_monitor/src/planning_scene_monitor.cpp:498

Waiting to continue: Press 'next' in the RvizVisualToolsGui window to start the demo
==========================================================================

Why this happened? and when I push the 'next' button

this error came out.
=========================================================================

[ERROR] [1542094584.358433716]: Found empty JointState message
[ INFO] [1542094584.358625544]: The timeout for planning must be positive (0.000000 specified). Assuming one second instead.
[ WARN] [1542094584.358758867]: It looks like the planning volume was not specified.
[ WARN] [1542094584.359060750]: Returning dirty link transforms
[ WARN] [1542094584.359205771]: Returning dirty link transforms
[ WARN] [1542094584.359819955]: Returning dirty link transforms
[ WARN] [1542094584.359895225]: Returning dirty link transforms
[ INFO] [1542094584.361217367]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ WARN] [1542094584.362594239]: RRTConnect: Skipping invalid start state (invalid state)
[ERROR] [1542094584.362725271]: RRTConnect: Motion planning start tree could not be initialized!
[ INFO] [1542094584.362798699]: No solution found after 0.000639 seconds
[ WARN] [1542094584.371471109]: Goal sampling thread never did any work.
[ INFO] [1542094584.371718926]: Unable to solve the planning problem
[ERROR] [1542094584.371790032]: Could not compute plan successfully
[ WARN] [1542094584.382583638]: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.

=================================
What's the problem

labarea, 2018-12-04:
We are having the same problem... we tried to change the topic in rviz but it not works

AjSat, 2018-12-11:
I am having the same issue as well. Changing the topic as mentioned in #188 does not resolve the issue

liuxin00738, 2018-12-14:
Same issue here. I think it is the initial state or launch file config.

mlautman, 2018-12-14:
> [ WARN] [1542094584.362594239]: RRTConnect: Skipping invalid start state (invalid state)
[ERROR] [1542094584.362725271]: RRTConnect: Motion planning start tree could not be initialized!
[ INFO] [1542094584.362798699]: No solution found after 0.000639 seconds

The robot seems to be loading up in  an invalid start state. Can you tell us more about your setup? Are you building moveit_tutorials from source? Which panda_moveit_config repo are you using? Can you post a screen grab of rviz with the panda in it's starting configuration?

labarea, 2018-12-17:
Hi, we used 
git clone -b kinetic-devel https://github.com/ros-planning/moveit_tutorials.git
git clone -b kinetic-devel https://github.com/ros-planning/panda_moveit_config.git
from this link [http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/getting_started/getting_started.html](url) 
The initial configuration in rviz is this
[image]

mlautman, 2018-12-22:
Thanks for that information. I just opened a PR fixing the bug and quite a few others

closed 2018-12-22 by mlautman

HDallali, 2019-01-02:
Hi, I'm having the same problem. is there a fix for it?

warpdriv, 2020-06-22:
Hi, I'm just getting into ROS and am having the same problem. When will the bug fix be applied to the non-dev branch so others will not run into this issue? How can I get the fix from the devel branch?
Thank you.