moveit/moveit_tutorials #262: Error trying to follow melodic tutorial

Afladimir1, 2018-11-30:
I was following the tutorial of moveit for melodic and I was able to reach the part to configure your catkin workspace but once I type catkin build I get these errors.

`Errors     << moveit_tutorials:make /home/user/ws_moveit/logs/moveit_tutorials/build.make.001.log
In file included from /usr/include/eigen3/Eigen/Core:347:0,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /opt/ros/melodic/include/moveit/robot_model/joint_model.h:47,
                 from /opt/ros/melodic/include/moveit/robot_model/joint_model_group.h:41,
                 from /opt/ros/melodic/include/moveit/robot_model/robot_model.h:47,
                 from /opt/ros/melodic/include/moveit/robot_model_loader/robot_model_loader.h:41,
                 from /home/user/ws_moveit/src/moveit_tutorials/doc/robot_model_and_robot_state/src/robot_model_and_robot_state_tutorial.cpp:40:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/user/ws_moveit/src/moveit_tutorials/doc/robot_model_and_robot_state/src/robot_model_and_robot_state_tutorial.cpp:118:102:   required from here
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:331:5: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
In file included from /usr/include/eigen3/Eigen/Core:347:0,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /opt/ros/melodic/include/moveit/robot_model/joint_model.h:47,
                 from /opt/ros/melodic/include/moveit/robot_model/joint_model_group.h:41,
                 from /opt/ros/melodic/include/moveit/robot_model/robot_model.h:47,
                 from /opt/ros/melodic/include/moveit/robot_model_loader/robot_model_loader.h:41,
                 from /home/user/ws_moveit/src/moveit_tutorials/doc/state_display/src/state_display_tutorial.cpp:40:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/user/ws_moveit/src/moveit_tutorials/doc/state_display/src/state_display_tutorial.cpp:101:114:   required from here
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:331:5: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
In file included from /usr/include/eigen3/Eigen/Core:347:0,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /opt/ros/melodic/include/moveit/robot_model/joint_model.h:47,
                 from /opt/ros/melodic/include/moveit/robot_model/joint_model_group.h:41,
                 from /opt/ros/melodic/include/moveit/robot_model/robot_model.h:47,
                 from /opt/ros/melodic/include/moveit/robot_model_loader/robot_model_loader.h:41,
                 from /home/user/ws_moveit/src/moveit_tutorials/doc/interactivity/include/interactivity/interactive_robot.h:43,
                 from /home/user/ws_moveit/src/moveit_tutorials/doc/interactivity/src/interactive_robot.cpp:39:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/user/ws_moveit/src/moveit_tutorials/doc/interactivity/src/interactive_robot.cpp:93:79:   required from here
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:331:5: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
make[2]: *** [doc/robot_model_and_robot_state/CMakeFiles/robot_model_and_robot_state_tutorial.dir/src/robot_model_and_robot_state_tutorial.cpp.o] Error 1
make[1]: *** [doc/robot_model_and_robot_state/CMakeFiles/robot_model_and_robot_state_tutorial.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [doc/state_display/CMakeFiles/state_display_tutorial.dir/src/state_display_tutorial.cpp.o] Error 1
make[1]: *** [doc/state_display/CMakeFiles/state_display_tutorial.dir/all] Error 2
make[2]: *** [doc/interactivity/CMakeFiles/interactivity_utils.dir/src/interactive_robot.cpp.o] Error 1
make[1]: *** [doc/interactivity/CMakeFiles/interactivity_utils.dir/all] Error 2
In file included from /usr/include/pcl-1.8/pcl/sample_consensus/sac_model.h:52:0,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/sac.h:45,
                 from /usr/include/pcl-1.8/pcl/segmentation/sac_segmentation.h:49,
                 from /home/user/ws_moveit/src/moveit_tutorials/doc/perception_pipeline/src/cylinder_segment.cpp:43:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h: In function ‘void __static_initialization_and_destruction_0(int, int)’:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: warning: ‘pcl::SAC_SAMPLE_SIZE’ is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
   SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
   ^~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: note: declared here
make: *** [all] Error 2
`

jarvisschultz, 2018-12-06:
It looks like your error is related to [this commit](https://github.com/ros-planning/moveit/commit/ea7125e72e556b8bc74f0d5e542b95a8a27c6b0c#diff-22ed9d5e496d7469a03e0b414299ebde) in `moveit` which switched many functions that return `Eigen::Affine3d` to use `Eigen::Isometry3d`. Correspondingly, this repo updated to use `Eigen::Isometry3d` on [this commit](https://github.com/ros-planning/moveit_tutorials/commit/b6af4487e57acc1b70a3c4f7bbd58e53f23d32a0). Your issue is almost certainly that you have checked out a branch of  `moveit_tutorials`  that is a child of the previously-linked commit (thus including the `Eigen::Isometry3d` change), but you are building against the moveit Debian package that has not yet incorporated this change. Two possible fixes:

- Build MoveIt! from source in your workspace from the `melodic-devel` branch
- Check out an earlier commit on the `moveit_tutorials` repo

jwhendy, 2019-01-19:
I'm getting a similar error on `moveit` itself.

```
### fresh clone from today
$ git log
abc4f4070e300352c851f84fb53fa6d06c9eacb

Errors     << moveit_core:make /home/jwhendy/catkin_ws/logs/moveit_core/build.make.000.log                                                                                               
/home/jwhendy/catkin_ws/src/moveit/moveit_core/robot_state/src/robot_state.cpp: In member function ‘void moveit::core::RobotState::copyFrom(const moveit::core::RobotState&)’:
/home/jwhendy/catkin_ws/src/moveit/moveit_core/robot_state/src/robot_state.cpp:161:79: warning: ‘void* memcpy(void*, const void*, size_t)’ writing to an object of type ‘Eigen::Isometry3 ’ {aka ‘class Eigen::Transform<double, 3, 1>’} with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
     memcpy(variable_joint_transforms_, other.variable_joint_transforms_, bytes);
                                                                               ^
In file included from /usr/include/eigen3/Eigen/Geometry:44,
                 from /home/jwhendy/catkin_ws/src/moveit/moveit_core/robot_model/include/moveit/robot_model/joint_model.h:47,
                 from /home/jwhendy/catkin_ws/src/moveit/moveit_core/robot_model/include/moveit/robot_model/joint_model_group.h:41,
                 from /home/jwhendy/catkin_ws/src/moveit/moveit_core/robot_model/include/moveit/robot_model/robot_model.h:47,
                 from /home/jwhendy/catkin_ws/src/moveit/moveit_core/robot_state/include/moveit/robot_state/robot_state.h:41,
                 from /home/jwhendy/catkin_ws/src/moveit/moveit_core/robot_state/src/robot_state.cpp:38:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:201:7: note: ‘Eigen::Isometry3d’ {aka ‘class Eigen::Transform<double, 3, 1>’} declared here
 class Transform
       ^~~~~~~~~
In file included from /usr/include/eigen3/Eigen/Core:366,
                 from /opt/ros/melodic/include/eigen_stl_containers/eigen_stl_vector_container.h:40,
                 from /opt/ros/melodic/include/eigen_stl_containers/eigen_stl_containers.h:40,
                 from /opt/ros/melodic/include/geometric_shapes/bodies.h:45,
                 from /home/jwhendy/catkin_ws/src/moveit/moveit_core/collision_distance_field/include/moveit/collision_distance_field/collision_distance_field_types.h:48,
                 from /home/jwhendy/catkin_ws/src/moveit/moveit_core/collision_distance_field/src/collision_distance_field_types.cpp:37:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/jwhendy/catkin_ws/src/moveit/moveit_core/collision_distance_field/src/collision_distance_field_types.cpp:56:56:   required from here
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:331:25: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
                         ^~~~~~~~~~~~~
make[2]: *** [collision_distance_field/CMakeFiles/moveit_collision_distance_field.dir/build.make:63: collision_distance_field/CMakeFiles/moveit_collision_distance_field.dir/src/collision_distance_field_types.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:6594: collision_distance_field/CMakeFiles/moveit_collision_distance_field.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
cd /home/jwhendy/catkin_ws/build/moveit_core; catkin build --get-env moveit_core | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
```

I realize that this is not explicitly about the tutorials, but since similar posts on ROS Answers also link here and this is the top hit for the `Eigen::Isometry3d` error, I thought I'd piggy back. I can open a new error if that's preferred.

From the above, the error seemed to be that the tutorials needed an up to date MoveIt, but that's where I'm stuck.

jwhendy, 2019-01-20:
This is solved for me after finding [this issue](https://github.com/ros-planning/moveit/pull/1096). I'm on arch linux, so instead of using the `ros-melodic-*` AUR offerings, I cloned the following and built the following manually from `catkin_ws/src`: geometric_shapes, geometry2, moveit, moveit_tutorials, moveit_visual_tools, and rviz_visual_tools.

Looks like these changes will propagate soon. The main thing I'd add that wasn't clear from @jarvisschultz 's comment above is that you need to build more than just MoveIt to get this all to play nicely.

closed 2019-01-20 by Afladimir1

jarvisschultz, 2019-01-22:
If you build MoveIt! from source using [the official instructions](https://moveit.ros.org/install/source/), you should automatically get all packages @jwhendy mentioned cloning manually.

jwhendy, 2019-01-22:
@jarvisschultz Good point. I was on the path of using AUR, which is based on the split out packages at [ros-release](https://github.com/ros-gbp/ros-release) and reflexively looked pulled things in on a per-package basis when my builds failed. Thanks for the correction.

jarvisschultz, 2019-01-22:
@jwhendy Not a correction at all :) Just wanted to make sure anyone reading this thread had the full story. You're totally right about this report showing up on answers.ros.org and Google results.

MrDadaGuy, 2019-02-11:
Ubuntu 18.04, ROS Melodic...  I followed the build from source instructions, but I was still having this error:

> /home/modsim/ws_moveit/src/moveo_ros/moveo_moveit/src/move_group_interface_coor_1.cpp:65:95:   required from here
> /usr/include/eigen3/Eigen/src/Geometry/Transform.h:331:5: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
>      EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
> 

I modified moveo_ros/moveo_moveit/src/move_group_interface_coor_1.cpp as follows:

```
//  Eigen::Affine3d text_pose = Eigen::Affine3d::Identity();  <-- comment this out, replace with below line
  Eigen::Isometry3d text_pose = Eigen::Isometry3d::Identity();
```

Now I'm able to build.   Putting this here because this was the only search result for my query, in hopes it can help someone else.  Thanks,

jarvisschultz, 2019-02-11:
@MrDadaGuy this looks like a bug with a completely different package. Perhaps you should consider opening a bug report or PR with `moveo_ros`

magnuspyke, 2019-03-12:
I ran into the same issue @MrDadaGuy ran into while compiling [jesseweisberg/moveo_ros](https://github.com/jesseweisberg/moveo_ros) on Ubuntu 18.04 with ROS Melodic.  Applying the change to moveo_ros/moveo_moveit/src/move_group_interface_coor_1.cpp given by @MrDadaGuy also allowed me to build without errors.  I will go ahead and add an issue on https://github.com/jesseweisberg/moveo_ros.