moveit/moveit_tutorials #265: Releasing moveit_tutorials?

rhaschke, 2018-12-10:
I'm not sure why, but we never actually released moveit_tutorials into Indigo or later:
https://github.com/ros-planning/moveit_tutorials/pull/64#issuecomment-282568138

Before this, back in Hydro, there was a pr2_moveit_tutorials.
@v4hn, @davetcoleman Any objections to create a binary release?

davetcoleman, 2018-12-10:
we've had this discussion several times, see [my response last time](https://github.com/ros-planning/moveit_tutorials/pull/64#issuecomment-285907307)

Overall, what do you hope to gain from releasing it? There's nothing useful in a debian package to a user learning moveit... all they may want to see is the source code so they can run it locally and modify it.

rhaschke, 2018-12-10:
Users just might want to run the tutorial examples. Looking at the source code is only the 2nd step.
I don't have a very strong opinion on that, just wondered because many other ROS tutorials are available as binary packages:

ros-melodic-actionlib-tutorials
ros-melodic-common-tutorials
ros-melodic-geometry-tutorials
ros-melodic-interactive-marker-tutorials
ros-melodic-librviz-tutorial
ros-melodic-nodelet-tutorial-math
ros-melodic-pluginlib-tutorials
ros-melodic-ros-tutorials
ros-melodic-roscpp-tutorials
ros-melodic-rospy-tutorials
ros-melodic-rviz-plugin-tutorials
ros-melodic-rviz-python-tutorial
ros-melodic-urdf-sim-tutorial
ros-melodic-urdf-tutorial
ros-melodic-visualization-marker-tutorials
ros-melodic-visualization-tutorials

As a teacher, these binary repos are always handy for me, because - if they are installed in the system - I can immediately showcase their examples. Actually, it's not a big deal to release this repo as well.

davetcoleman, 2018-12-10:
Seems to me just like more release overhead and very little gain. If you want to showcase moveit just run demo.launch or the MSA

I don't have a strong opinion other than... why bother?

davetcoleman, 2018-12-10:
e.g. is a user actually asking for this? other than the ones who are completely confused about everything?

closed 2019-01-18 by rhaschke