moveit/moveit_tutorials #280: Merge pull-requests by commit disabled?

v4hn, 2019-01-21:
I just wanted to merge #278 and found the `Create a merge commit` field disabled:

> Create a merge commit
> All commits from this branch will be added to the base branch via a merge commit.
> Not enabled for this repository

@davetcoleman @mlautman @rhaschke Did someone disable this option?
I guess it is still possible to do this by hand in git, right?
It's not a big deal for this request, but in general I would prefer to have this possibility.

rhaschke, 2019-01-22:
The citation primarily refers to single commits and fixup commits, which indeed should not clutter the git history. Merge-commits make a lot of sense if there are multiple, clearly distinguishable commits in a single PR!

davetcoleman, 2019-01-22:
For clearly distinguishable commits in a single PR we've all agreed we want *rebase merging*:

[image]

rhaschke, 2019-01-22:
Often that's a viable option, yes. However, rebase-merging also disguises the extension of a merge (you don't see the start and end of the related commits anymore) and the PR id is not reported by github.
For this reason, I prefer merge commits in other projects, although I try not to merge branches that were branched off very deep in the past. In this case, I first rebase and then merge-commit.

v4hn, 2019-01-24:
> For clearly distinguishable commits in a single PR we've all agreed we want rebase merging:

I'm not aware of an agreement to *never* use merge commits in moveit and neither Robert nor I seem to agree.

- single commits - > squash
- multiple fixup-style commits - > squash
- multiple independent commits -> rebase and merge
- multiple clean commits with common ground and some relation - > merge commit.

one of git's design goals are short-living feature branches that can be merged.
So I really don't think it makes sense to disable merges...

davetcoleman, 2019-01-25:
Sounds good, I've re-enabled merging. Can you copy that policy your wrote out on our Pull Request page please? You can just edit on Github:
https://github.com/ros-planning/moveit.ros.org/blob/gh-pages/documentation/contributing/pullrequests/index.markdown

closed 2019-02-28 by mlautman