moveit/moveit_tutorials #284: Deserialization error when get_current_state()

hjzh4, 2019-02-18:
### Description

I was trying python api tutorial. When it executed robot.get_current_state(), I got TypeError: a bytes-like object is required, not 'str'

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source build
* If source, which git commit or tag? kinetic-devel

### Backtrace or Console output
Traceback (most recent call last):
  File "/home/sawyer_docker/pycharm-community-2018.3.4/helpers/pydev/pydevd.py", line 1741, in <module>
    main()
  File "/home/sawyer_docker/pycharm-community-2018.3.4/helpers/pydev/pydevd.py", line 1735, in main
    globals = debugger.run(setup['file'], None, None, is_module)
  File "/home/sawyer_docker/pycharm-community-2018.3.4/helpers/pydev/pydevd.py", line 1135, in run
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "/home/sawyer_docker/pycharm-community-2018.3.4/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/home/sawyer_docker/modelbasedsim2real/moveit_test.py", line 36, in <module>
    print(robot.get_current_state())
  File "/home/sawyer_docker/ros_ws/src/moveit/moveit_commander/src/moveit_commander/robot.py", line 227, in get_current_state
    s.deserialize(self._r.get_current_state())
  File "/home/sawyer_docker/ros_ws/devel/lib/python3/dist-packages/moveit_msgs/msg/_RobotState.py", line 627, in deserialize
    (_x.joint_state.header.seq, _x.joint_state.header.stamp.secs, _x.joint_state.header.stamp.nsecs,) = _get_struct_3I().unpack(str[start:end])
TypeError: a bytes-like object is required, not 'str'

\\

mlautman, 2019-02-22:
For one thing you should not be using Pyhon3 with Kinetic on 16.04.

Closing this issue since it isn't a bug with the tutorials. A better place to bring this up would be [answers.ros.org](answers.ros.org) or at least on the moveit repo itself.

closed 2019-02-22 by mlautman

davetcoleman, 2019-02-23:
@hjzh4 thanks for reporting, let us know if you do find an answer though!

@mlautman is the lack of python3 support well documented for the kinetic tutorials?

hjzh4, 2019-02-24:
@davetcoleman 
Hi Dave, thanks for replying. I know that python3 is not  officially supporting by moveit (or even ROS :p). However, I did compile moveit with python3 interpreter successfully half year ago and it just doesn't work now ( .. ). Currently I'm using python2 while I hope moveit can start supporting python3 in the future as many packages in RL community only have python3 versions.