moveit/moveit_tutorials #290: Status of tutorials on Windows?

imcardle, 2019-02-28:
### Description

I'm new to ROS and MoveIt! so I have been following the tutorials for Windows here:
[ROSOnWindows - MoveIt Tutorials](https://ms-iot.github.io/ROSOnWindows/Moveit/moveit_tutorials.html)

However, I'm running into problems with rViz not starting up properly (error message "Entry Point Not Found" when trying to call robot_state_publisher.exe).

Can anyone offer a workaround or solution for this issue please?

### Your environment
* ROS Distro: Melodic
* OS Version: Windows 10
* Binary build - where do I find this information?  I upgraded to the latest versions of ROS and MoveIt! today using choco upgrade.

### Steps to reproduce
Execute the following tutorial demo:
_roslaunch panda_moveit_config demo.launch_

### Expected behaviour
rViz should start up without any error message.

### Backtrace or Console output
[image]

mlautman, 2019-02-28:
@imcardle MoveIt! on Windows isn't currently supported and is definitely not recommended. Perhaps ping the MS-IOT team at https://github.com/ms-iot/moveit_tutorials

closed 2019-02-28 by mlautman

davetcoleman, 2019-03-06:
@seanyen-msft

seanyen-msft, 2019-03-06:
Thank @davetcoleman bringing my attention on this. @imcardle, if it still an issue, can you open it on https://github.com/ms-iot/rosonwindows/issues? Our team will take a look on that. Btw, you can try to do "choco upgrade ros-melodic-moveit -y" to refresh the bits in case you are unluckily getting a broken build.

imcardle, 2019-03-06:
Hi @seanyen-msft - I updated today, so I'm now running build 20190305.1 but the issue still persists.  I've raised a new issue on github as you suggested:
[https://github.com/ms-iot/ROSOnWindows/issues/82](https://github.com/ms-iot/ROSOnWindows/issues/82)