moveit/moveit_tutorials #296: [Help] the difference between "Motion planning API" "Motion planning pipeline" "Move group Interface"

ghx0x0, 2019-03-14:
hello,
I'm a freshman in moveit. I read and tried the moveit toturials, and seems I can plan trajectories in 3 ways, using motion planning API/ motion planning pipeline/ Move group:

planning_interface::PlanningContextPtr context;
context->solve(res);
planning_pipeline->generatePlan(planning_scene, req, res);
moveit::planning_interface::MoveGroupInterface::Plan my_plan;
move_group.plan(my_plan) ;
but, can anybody tell me the difference between them?

mlautman, 2019-03-20:
There are multiple ways to do almost everything in MoveIt. The best way to think of MoveIt is a collection of really useful components that you can mix and match as needed. The MoveGroup is designed to abstract away as much as possible by giving the user a unified interface for doing a lot of the most common tasks. The advantage of using the MoveGroup is that it is really easy to get started. The downside is that it isn't easy to customize. For anything complex, it is best to use MoveIt's components to achieve your ends. The motion planning api is a C++ interface for motion planning given a planning scene and some constraints. The motion planning pipeline is useful if you want to add additional processing steps like timing parameterization or trajectory smoothing.

mlautman, 2019-03-20:
In the future please post questions to answers.ros.org and save issues for bug reports

closed 2019-03-20 by mlautman

ghx0x0, 2019-04-12:
> There are multiple ways to do almost everything in MoveIt. The best way to think of MoveIt is a collection of really useful components that you can mix and match as needed. The MoveGroup is designed [...]

Hi, mlautman. Thank you so much for your quick response. That's very helpful.