moveit/moveit_tutorials #297: Missing interactive markers

imcardle, 2019-03-14:
### Description

I'm running through the MoveIt! getting started tutorial but I'm not able to see the "interactive markers" in rViz. I've clicked on the "Interact" button, as suggested in the tutorial - but still the markers don't show up.

### Your environment
* ROS Distro: [Indigo|Jade|Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

ROS Distro: Melodic
OS Version: Ubuntu 18.04 (running in VirtualBox)
Binary build:  Where can I find this information?
RViz: 1.13.1
Qt: 5.9.5
OGRE: 1.9.0

### Steps to reproduce
Follow the tutorial steps here:
[https://ros-planning.github.io/moveit_tutorials/](https://ros-planning.github.io/moveit_tutorials/)

### Expected behaviour
Interactive markers should appear in the 3D view of rViz when the "Query Start State" and "Query Goal State" options are ticked, and when the "Interact" toolbar button is pressed.

### Backtrace or Console output

The console output includes the following warnings - which may or may not be related to this issue:
[ WARN] [1552568009.202946440]: Unable to update multi-DOF joint 'virtual_joint': Failure to lookup transform between 'world' and 'panda_link0' with TF exception: "world" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1552568009.499416675]: The complete state of the robot is not yet known.  Missing virtual_joint

imcardle, 2019-03-15:
Just realised that this repo is for the tutorial documentation, rather than the tutorial code.  Moved the issue to the ros-planning/moveit repo instead.

closed 2019-03-15 by imcardle

mlautman, 2019-03-20:
I opened a PR that will fix the issue. Thanks for bringing this up with us!