moveit/moveit_tutorials #300: Rename `planner` to pipeline in launch examples

jschleicher, 2019-03-21:
### Description

[moveit#1397](https://github.com/ros-planning/moveit/pull/1397/files) shows, that [the tutorials use `planner`](https://ros-planning.github.io/moveit_tutorials/doc/chomp_planner/chomp_planner_tutorial.html#using-chomp-with-your-robot) for the planning pipeline while the setup assistant generates launch files using `pipeline` arguments. The tutorials should be updated to reflect the same names as in the setup_assistant launch files.

This should also update the [panda_moveit_config` package](https://github.com/ros-planning/panda_moveit_config/blob/melodic-devel/launch/move_group.launch) to be consistant with the tutorial page that references it.

davetcoleman, 2019-03-26:
Once your https://github.com/ros-planning/moveit/pull/1397 is merged in, please make this update @jschleicher

jschleicher, 2019-03-27:
As said in #1397: I'm not a panda expert; updating the tutorials should also re-generate the [`panda_moveit_config`](https://github.com/ros-planning/panda_moveit_config/blob/melodic-devel/launch/move_group.launch). I'm not feeling confident about making this change.